whatwg/bluetooth #68: startNotifications using Notification or Indication?

scheib, 2015-02-09:
Currently startNotifications / [enable notification](http://webbluetoothcg.github.io/web-bluetooth/#dfn-enable-notifications) leaves room for Notifications or Indications being used. 

> Use any of the Characteristic Descriptors procedures to ensure that one of the Notification or Indication bits in characteristic's Client Characteristic Configuration descriptor is set, matching the constraints in characteristic's properties. The UA should avoid setting both bits, and must deduplicate value-change events if both bits are set. 

I'm not clear yet what would cause the UA to attempt to set both bits, and why 'should' is used instead of 'must'. 

Presuming either may be set, UAs will need to determine which to use. however, this is likely service and application dependent. I believe it may be prudent to expose both Notification and Indication to the Web Bluetooth API. Or, failing that to use another name such as startUpdates or startSignals and to specify the preferred order of Notification or Indication.

jyasskin, 2015-02-09:
Not all of the platform APIs let the application pick which of Notifications or Indications to use, so we can't insist that web apps be able to or say what the preferred order is, or make UAs guarantee that their underlying platform doesn't set both. I think we have the strongest requirement we can.

scheib, 2015-02-09:
Ah, I see. I also see that 'Notification' terminology is used to abstract over Notification and Indication. 

https://developer.apple.com/library/ios/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/PerformingCommonCentralRoleTasks/PerformingCommonCentralRoleTasks.html

https://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#setCharacteristicNotification(android.bluetooth.BluetoothGattCharacteristic, boolean)

closed 2015-02-09 by scheib