whatwg/bluetooth #75: Add code examples to the spec

jyasskin, 2015-03-16:
@slightlyoff requested either code examples in the spec or a prominent link to the [explainer](https://github.com/WebBluetoothCG/web-bluetooth/blob/gh-pages/explainer.md). Code examples are probably better for the spec itself to avoid versioning issues.

jyasskin, 2015-06-17:
I'm going to remove the [explainer](https://github.com/WebBluetoothCG/web-bluetooth/blob/98e2c1648a418c9ef3c6d9b535130225e8aaca08/explainer.md) for being out of date, so here's a list of things it intended to document that we don't have code examples for yet:

* Add a device using a non-standard UUID to show off the non-string interface.
* Show how to get all instances of a given service UUID, for example to support multiple battery services in a device.
* Show an example of querying a service's UUID to motivate the bluetoothUUID.get*() functions.
* [Not spec'ed yet] Register to get notified when a particular device comes in range. UA would present user with a dialog asking "which app do you want to handle this device?"
* Write a characteristic
* Read/write a descriptor

The existing examples could use more exposition, too.

beaufortfrancois, 2016-01-06:
I believe we can close this one. @jyasskin WDYT?

jyasskin, 2016-02-02:
I still don't have all the examples from my old comment. :-/