whatwg/bluetooth #78: Design Peripheral/GATT Server support

jyasskin, 2015-03-20:
Split from #22. I have an initial sketch for this at https://docs.google.com/document/d/1BKBY39CeUWzkmEExTml0JUEy2eHD4upixt1_7zqjMkg/edit?usp=sharing, which is open to comments. It's not a priority for the first release, but we can collect ideas here so they don't get lost.

jyasskin, 2015-05-29:
FirefoxOS is working on a server API at https://groups.google.com/d/topic/mozilla.dev.webapi/OCkEhkpwqVk/discussion / https://wiki.mozilla.org/B2G/Bluetooth/WebBluetooth-v2/BluetoothGattServer#BluetoothGattServer.

martijnthe, 2017-03-14:
What's the status on Web Bluetooth peripheral / GATT server support?

JFYI, I'm seeing JS APIs being designed for this in the IoT/device community.
For example, a couple existing JS BLE APIs:
- Zephyr.js based their JS BLE peripheral API on bleno: https://github.com/01org/zephyr.js/blob/master/docs/ble.md
- Another JS BLE peripheral API (limited to basic advertising):
https://github.com/Samsung/iotjs/blob/master/docs/api/IoT.js-API-BLE.md

jyasskin, 2017-03-14:
We're planning to do the scanning API first. After that, we'll take any existing peripheral/server APIs, especially ones linked here, into account when designing the web API.

jracle, 2019-07-26:
Voting strongly for this feature!

josephrocca, 2021-06-11:
Looking forward to this. My specific use case is to allow users (via a web app) to use their phone as a microphone for their laptop or other device. There are of course more details, but that's the 1 sentence summary of the relevant part. If I understand correctly, that would be possible with this feature. If any of the pros here have roadmap estimates and/or details on what's happening in this space across the various implementers and relevant committees, I'd love to hear those!

Weedshaker, 2026-09-23:
This is stalled since ever but peer-to-peer connection is as important than never before. Please, anyone... give us an update?