whatwg/bluetooth #80: Physical web and manifest

kenchris, 2015-03-23:
Currently the physical web is said to transmit a URL. Would it make sense to transmit JSON instead with a start_url as defined in http://w3c.github.io/manifest/ ? That way we can get a more web app like experience. Another option would be to transmit the URL referring to a manifest.

@marcoscaceres @anssik @jyasskin

jyasskin, 2015-03-23:
You may want to file this in one of the Physical Web's repositories, either https://github.com/google/physical-web or https://github.com/google/uribeacon.

The physical web's data has to fit in an [advertising packet](https://github.com/google/uribeacon/blob/master/specification/AdvertisingMode.md), so they only get 17 bytes. Even most URLs don't fit, let alone anything more complicated. The URL can, of course, refer to a web page that links to a manifest.

@scottjenson

kenchris, 2015-03-23:
I can do that. I filed it here as it was in the use-case document. But let me file it there as well

kenchris, 2015-03-23:
Filed: https://github.com/google/physical-web/issues/301

closed 2015-04-02 by jyasskin