whatwg/bluetooth #81: Clarify the "smart"

kenchris, 2015-03-23:
mode known as "Smart"

I would say "marketed as Bluetooth Smart. Devices supporting both BTLE and Classic Bluetooth, are marketed under the name Bluetooth Smart Ready."

jyasskin, 2015-03-23:
I don't really want to get into a long discussion of the subtleties of Bluetooth's marketing in the introduction of the spec. Should I say "Bluetooth Smart" instead of just "Smart"? The Bluetooth SIG [refers to both "Bluetooth Smart Ready" and "Bluetooth Smart" as the "Bluetooth® Smart marks"](http://www.bluetooth.com/pages/smart-logos-FAQ.aspx), so I think I'm on safe ground grouping them under that name. I don't think the distinction between "known as" and "marketed as" really matters here—if anything, "marketed as" is derogatory which I'd like to avoid.

kenchris, 2015-03-23:
I would say "Bluetooth Smart", that is clearer

commit 24c13fb9 (2015-03-23): Fix #81 by using "Bluetooth Smart" instead of just "Smart"

This matches the Bluetooth SIG's documents better.
closed 2015-03-23 by jyasskin