whatwg/bluetooth #83: Confusing uppercase/lowercase

kenchris, 2015-03-23:
ID vs uuid, e.g. vendorID, uuids, BluetoothServiceUuid, connectGATT

I know the web platform itself is pretty inconsistent here. @marcoscaceres do you know if there is some kind of agreement on casing for new APIs? We need the same for NFC. @zolkis

marcoscaceres, 2015-03-25:
I don't know, tbh :( However, saying what the type represents doesn't always add value (because, for instance, generally you can't do anything with the fact that something is a UUID). So, if anything, maybe see if you can simplify the property names to what they are or role they play (i.e., just `.id`, instead of `.uuid`).

jyasskin, 2015-04-08:
@scheib IIRC you pointed out a precedent for either all-caps'ing acronyms or camel-casing them. Which was it?

@marcoscaceres `.id` isn't enough because we have 2 kinds of IDs for services/characteristics/descriptors: the UUID that identifies the meaning of the s/c/d, and the instance ID that distinguishes multiple s/c/d's with the same UUID. We could make up a term like 'type' for the "meaning" UUID, but I think that would be confusing to people who've read about Bluetooth programming elsewhere.

marcoscaceres, 2015-04-08:
@jyasskin thanks for the clarification. I don't have a strong opinion.

scheib, 2015-04-15:
I made changes in the past changing case to all caps: e.g. instanceId -> instanceID. This was based on some quick spec searching and asking a few Chrome folks - it wasn't a obvious or clean decision, but I like it.

I think we should move all other uses to all caps, e.g. BluetoothServiceUuid -> BluetoothServiceUUID.

.uuid appears as an attribute on several interfaces, e.g. BluetoothDevice. Tricky, but I think we should go with the lowercase leading portions of attribute names convention, and thus leave these as .uuid.

If it were a multi term attribute leading with an acronym, I'd suggest uuidOtherWords.

kenchris, 2015-04-16:
I think that makes sense and is consistent with the current [not always too consistent :-)] web platform

commit 8154c44f (2015-04-20): All-caps acronyms except when they occur at the start of a name.

Thus "BluetoothDescriptorUUID" but "txPower".

Fixes #83.
closed 2015-04-20 by jyasskin