whatwg/bluetooth #88: The API description in the Use Cases doesn't match the one in the Spec

g-ortuno, 2015-03-30:
Spec description: `API to discover and communicate with devices over the Bluetooth 4 wireless standard using the Generic Attribute Profile (GATT).`

Use cases description: `... interface to discover and communicate with devices connected via Bluetooth to the device hosting the implementation.`

jyasskin, 2015-03-30:
I'm not sure the spec's description should match the use-cases description? The use cases and security requirements came first and drove the decisions in the spec, including the decision to limit to GATT.

g-ortuno, 2015-03-30:
hmm ok. But I still think the wording in the Use Cases' API description is a bit weird. So I thought the best solution would be to match the spec's description.

jyasskin, 2015-03-30:
Would it make sense to replace the declarative "The Web Bluetooth API defines an interface to …" with something like "We intend to define an API to …"?

g-ortuno, 2015-03-30:
So: `We intend to define an API that uses Bluetooth to discover and communicate with devices. This document collects use cases for such an API.`

jyasskin, 2015-03-30:
Sure, that looks good.

PR #86 (2015-03-30): Changed the description of Web Bluetooth API to match Spec
Fixes #88
commit 4e138ad8 (2015-03-30): Changed wording of Introduction. Fixes #88.
closed 2015-03-30 by jyasskin