whatwg/bluetooth #91: Typo in third paragraph of Introduction.

g-ortuno, 2015-03-30:
I believe there is a typo in the second sentence of the third paragraph of the spec's [introduction](https://webbluetoothcg.github.io/web-bluetooth/#introduction):

`A Service exposes a collection of included Services and Characteristics.`

As far as I know a `Service` doesn't expose other `Services`. I believe it should read: `A Server exposes...`

jyasskin, 2015-03-30:
The Service exposes included Services through the `Find Included Services` sub-procedure. It's true that the client has to read those services by sending the Server commands, but that's true of characteristics and descriptors too, so I think it makes sense to talk about the Services exposing its included services.

g-ortuno, 2015-03-30:
Ahhh I see what you mean now. Shouldn't the spec briefly define what an `included Service` is then? Which if I understand correctly is just a reference to another `Service`. The current wording makes it seem that there are actually Services inside a Service.

jyasskin, 2015-03-30:
Yes, but maybe not in the Introduction. Now that https://webbluetoothcg.github.io/web-bluetooth/#information-model exists, we should probably elaborate in its Note about the Profile Hierarchy, and link to there from the Introduction. The Introduction is just trying to get people a really basic feel for the BTLE design so that the rest of the spec makes sense.

The particular user-visible thing that we need to say somewhere in the spec is that a single secondary service can be included by multiple other services. But we should also keep the logical sense described by BT4.2/3.G.2.6.3 where included services become "part of" the containing service's definition.

PR #115 (2015-05-22): Rewrite the introduction to shorten it and defer more to the BT spec.
Demo at https://rawgit.com/jyasskin/web-bluetooth-1/rewrite-intro/index.html.

The description of the GATT hierarchy moved to the Information Model section.

Fixes #91 by describing included services in more detail.
commit f74ff015 (2015-05-22): Rewrite the introduction to shorten it and defer more to the BT spec.

The description of the GATT hierarchy moved to the Information Model section.

Fixes #91 by describing included services in more detail.
closed 2015-05-22 by jyasskin