whatwg/bluetooth #124: getPrimaryService (and similar) should throw if UUID is not found

scheib, 2015-05-28:
After requestDevice succeeds for my set of services, an application believes it has access to a device that supports known services. Those services have known characteristics. Calling getPrimaryService(uuid1) and getCharacteristics(uuid2) should only have a promise fulfilled if services and characteristics existed and were obtained as expected. Applications shouldn't need to check for errors int he fulfillment of that promise:

```
navigator.bluetooth.requestDevicerequestDevice({
  filters: [ {services: [uuid1]} ]
}).then(function(device) {
  return device.connectGATT();
}).then(function(gattServer) {
  return gattServer.getPrimaryService(uuid1);
}).then(function(service) {
  return service.getCharacteristic(uuid2);
}).then(function(characteristics) {
  // Code here trusts that my requests have been fulfilled correctly.
  // If my uuids were correct, I know exactly the service, and that it
  // has certain number of characteristics with uuid2.
  // If for some reason they are not found, it is exceptional and
  // should have rejected a previous promise.
  characteristics[5]; // should exist and be valid without checking.
}).catch(... oh noes, normal exception handling ...);
```

- getCharacteristic requires a UUID, and it makes sense to reject if that uuid isn't found.
- getCharacteristics with a provided UUID should reject if not found for similar reasons.
- getCharacteristics without a provided UUID requires additional thought. I could go either way at the moment: I suppose if specific UUID isn't provided then the application doesn't have an expectation of what it will receive back, and an empty set may be valid. Consistency with "NotFound" == reject promise for the previous forms is reasonable as well.

jyasskin, 2015-05-29:
(You probably mean to call `service.getCharacteristics(uuid2)`, since [`getCharacteristic()`](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothgattservice-getcharacteristic) returns at most 1 object, not 6.)

If the developer calls `requestDevice(filter: [{services:[uuid1]}, {services:[uuid3]}])` then they need to call `getPrimaryService(uuid1)` to figure out which filter their device matched. Getting `null` back from that call isn't exceptional.

There are also cases, like in the [`heart_rate`](https://developer.bluetooth.org/gatt/services/Pages/ServiceViewer.aspx?u=org.bluetooth.service.heart_rate.xml) service, where certain characteristics are optional in the Service definition. It's not exceptional to request the `body_sensor_location` characteristic and get nothing back.

There are definitely other cases where the developer expects to always get a value back (`heart_rate_measurement`, for example), and the current design would produce a `TypeError: Cannot read property 'getCharacteristic' of null` if they didn't explicitly check the result.

We also have the chance that the device has moved out of range, in all of these cases. I wouldn't want developers to confuse that with an optional characteristic being absent.

I don't feel particularly strongly about this choice, but I suspect we should wait for developer feedback to finalize it.

scheib, 2015-05-29:
I'm undecided, but think for services/characteristics that I know may not be there I may still prefer to handle the scenario with rejected promises. Let's see what folks say when they try it.

Here's a rework the earlier code with current spec behavior. I suppose it's not so bad.:

```
navigator.bluetooth.requestDevicerequestDevice({
  filters: [ {services: [uuid1]} ]
}).then(function(device) {
  return device.connectGATT();
}).then(function(gattServer) {
  return gattServer.getPrimaryService(uuid1);
}).then(function(service) {
  if (!service)
    throw "service not found";
  return service.getCharacteristics(uuid2);
}).then(function(characteristics) {
  if (!characteristics[5])
    throw "characteristics[5] not found";
  characteristics[5]; // Ready to use it.
}).catch(... oh noes, normal exception handling ...);
```

beaufortfrancois, 2015-07-21:
My two cents: I'd rather see a rejected promise there than having to check if `service` exists since the goal of `getPrimaryService` is to actually get it.

beaufortfrancois, 2015-07-21:
Can we escalate this issue to https://github.com/WebBluetoothCG/web-bluetooth/labels/v1?

jyasskin, 2015-07-21:
@beaufortfrancois counts as a developer. Let's switch it, and see if folks complain.

closed 2015-07-24 by jyasskin

beaufortfrancois, 2015-09-17:
@g-ortuno Do we have a bug to track these changes?

g-ortuno, 2015-09-17:
There is now:

http://crbug.com/533108

PR #143 (2015-07-24): Throw NotFoundError when trying to look up a non-existent attribute.
The common case seems to be navigating a tree with certain attributes expected
to be there, and null checks make that more complicated. It's still possible to
handle optional attributes by catching the exception.

We have less experience with the multi-attribute getters, but I'm doing them
consistently for now.

Fixes #124.

Demo at https://rawgit.com/jyasskin/web-bluetooth-1/throw-for-missing-attributes/index.html. @beaufortfrancois @scheib
commit d95f6aa0 (2015-07-24): Throw NotFoundError when trying to look up a non-existent attribute.

The common case seems to be navigating a tree with certain attributes expected
to be there, and null checks make that more complicated. It's still possible to
handle optional attributes by catching the exception.

We have less experience with the multi-attribute getters, but I'm doing them
consistently for now.

Fixes #124.