whatwg/bluetooth #126: Helper functions to read values

g-ortuno, 2015-05-29:
In Android we have helper functions to more easily read and write values to and from characteristics:

[getFloatValue()](https://developer.android.com/reference/android/bluetooth/BluetoothGattCharacteristic.html#getFloatValue(int, int))
[getIntValue()](https://developer.android.com/reference/android/bluetooth/BluetoothGattCharacteristic.html#getIntValue(int, int))
[getStringValue()](https://developer.android.com/reference/android/bluetooth/BluetoothGattCharacteristic.html#getStringValue(int))

There are really useful since developers don't have to worry about bytes if all they are doing is reading/writing integers. Could we add similar helper functions?

jyasskin, 2015-05-29:
I think these features should be prototyped in a helper library before we put them in the spec, so they have a chance to evolve faster than they would in Blink. Shall we start a sibling repository to hold that helper library?

g-ortuno, 2015-05-29:
Eventually we should, for now let's focus on the spec and implementation.

beaufortfrancois, 2016-01-06:
I'd be happy to start this helper library.
Which other helper functions would you add?

beaufortfrancois, 2016-01-07:
@g-ortuno @jyasskin Before going further, is it what you were thinking of?
https://github.com/beaufortfrancois/sandbox/blob/99fd1453a12a1513f2731c9ffa561199294a9cc5/web-bluetooth/utils/utils.js

jyasskin, 2016-01-07:
Yes, that's great. I think I'd make 2 changes: I'd either default the `byteOffset` to `0` or omit it entirely, since if you're reading multiple items from a single characteristic, you probably want to only call `readValue` once. And I'd default `littleEndian` to `true`, since nearly all Bluetooth values are little-endian, contrary to `DataView`'s default.

beaufortfrancois, 2016-01-07:
Will do that tomorrow then. Thanks for the feedback.

beaufortfrancois, 2016-01-08:
I've changed default `littleEndian` to `true` as advised. Thank you!

Regarding the default/removal of `byteOffset`, that made me realized we might want to implement `BluetoothGattCharacteristic.prototype.getFloat32Value` instead of `BluetoothGattCharacteristic.prototype.readFloat32Value` so that it returns a Float32 value of the stored cached read value.
What do you think of adding an option to `readValue()` to return cached value such as 
```js
characteristic.readValue({cache: true})
.then()...
```

beaufortfrancois, 2016-01-08:
Here's what it would look like: https://github.com/beaufortfrancois/sandbox/blob/gh-pages/web-bluetooth/utils/utils.js

[image]

g-ortuno, 2016-01-11:
I wonder how useful these function will be when characteristic returns a
DataView instead of an ArrayBuffer.

beaufortfrancois, 2016-01-12:
Hopefully it won't be needed anymore.

beaufortfrancois, 2016-01-21:
Here's what it looks like at https://github.com/beaufortfrancois/sandbox/blob/gh-pages/web-bluetooth/utils/utils.js below. What do you think?

```js
BluetoothGATTCharacteristic.prototype.getFloat32Value = function(byteOffset, littleEndian = true) {
  return this.value.getFloat32(byteOffset, littleEndian);
};

BluetoothGATTCharacteristic.prototype.getFloat64Value = function(byteOffset, littleEndian = true) {
  return this.value.getFloat64(byteOffset, littleEndian);
};

BluetoothGATTCharacteristic.prototype.getInt16Value = function(byteOffset, littleEndian = true) {
  return this.value.getInt16(byteOffset, littleEndian);
};

BluetoothGATTCharacteristic.prototype.getInt32Value = function(byteOffset, littleEndian = true) {
  return this.value.getInt32(byteOffset, littleEndian);
};

BluetoothGATTCharacteristic.prototype.getInt8Value = function(byteOffset) {
  return this.value.getInt8(byteOffset);
};

BluetoothGATTCharacteristic.prototype.getStringValue = function(utfLabel = 'utf8') {
  var decoder = new TextDecoder(utfLabel);
  return decoder.decode(this.value);
};

BluetoothGATTCharacteristic.prototype.getUint16Value = function(byteOffset, littleEndian = true) {
  return this.value.getUint16(byteOffset, littleEndian);
};

BluetoothGATTCharacteristic.prototype.getUint32Value = function(byteOffset, littleEndian = true) {
  return this.value.getUint32(byteOffset, littleEndian);
};

BluetoothGATTCharacteristic.prototype.getUint8Value = function(byteOffset) {
  return this.value.getUint8(byteOffset);
};
```

jyasskin, 2016-05-04:
I'd be happy to have this as a simple npm module, but I'd like to wait to change the spec until we see it being widely used. I'll close this issue in this repository, but we can reopen later if it turns out people want this.

closed 2016-05-04 by jyasskin