whatwg/bluetooth #127: Say what to do when the system has no Bluetooth adapter or it's off.

jyasskin, 2015-06-02:
We need to specify whether `navigator.bluetooth` simply doesn't exist, or `navigator.bluetooth.requestDevice()` rejects its promise (with `NotSupportedError`?), or if it's detectable some other way (`navigator.bluetooth.enabled`?). If the adapter is turned off after the user gets a device, should that be a `NotSupportedError` or a `NetworkError` (simulating that the device went out of range)?

scheib, 2015-06-02:
The existence of the adapter and its powered state shouldn't be exposed until after device access is permitted by a user selecting a device. At that point I think `NetworkError` may make sense, but a more specific error type that indicates the adapter power is off would be cleaner for application developers to respond to. Apps would reasonably display UI to the effect of "To enable this, please turn on Bluetooth". 

navigator.bluetooth.enabled would be reasonable to query, except for the privacy concern. I suppose it could return `undefined` until the first device selected.

beaufortfrancois, 2015-07-08:
I believe `navigator.bluetooth.requestDevice` UA chooser should be the one responsible for communicating to the user the state of the Bluetooth Adapter:

- Bluetooth adapter doesn't exist. In that case rejects instantly properly the promise with some `BluetoothAdapterNotFoundError` error. Regarding privacy concern, it might be hard to avoid timing attacks there.

- Bluetooth adapter is off. "Turn on Bluetooth" should be available right there in the chooser if possible and would trigger a new BT search when enabled. 
If user dismisses it, the promise would be rejected with some `BluetoothAdapterNotFoundError` error.

mounirlamouri, 2015-07-08:
In ```navigator.bluetooth.requestDevice()``` going to require a user gesture of some sort?

beaufortfrancois, 2015-07-08:
TLDR; Yes.

According to https://webbluetoothcg.github.io/web-bluetooth/#the-ua-s-bluetooth-address, _This spec does require a user gesture for a website to trigger a scan, which reduces the frequency of scans some, but it would still be better for more platforms to expose the Privacy Feature._

jyasskin, 2015-07-08:
Do we want websites to be able to change their UI to avoid teasing users with a Bluetooth option they can't use? If so, we should have a way other than `requestDevice()` to learn that there's no adapter. (We should check with the privacy team before deciding that they'll tell us to hide this; we expose [WEBGL_debug_renderer_info](https://crbug.com/335373), for example.) Using `requestDevice()` seems fine to discover that the adapter's just disabled.

If we don't care about that UI, then having `requestDevice()` express everything makes sense.

Either way, we need to pick the error name to use. I've been avoiding adding new DOMException types to [WebIDL](https://heycam.github.io/webidl/#idl-DOMException-error-names), but we can do it if we want. I'll check with public-script-coord@w3.org.

jyasskin, 2015-07-08:
Actually, if the user has a disabled BLE adapter, and they decline to turn it on in the `requestDevice()` dialog, I think that should look the same as if they had an enabled adapter and either didn't find a matching device or clicked 'Cancel' to dismiss the dialog without selecting a device. That is, [reject with a `NotFoundError`](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetooth-requestdevice).

Is there a user story where it'd help for the site to be able to distinguish the three cases?

scheib, 2015-07-08:
User has no adapter: I'm OK with letting sites know ahead of time, but also
OK with showing the dialog and saying "Sorry, no adapter, can't do this."
and then reject the request.

User has an adapter turned off, declines to turn it on:
User has adapter turned on, declines to select device: Both just reject the
request.

jyasskin, 2015-07-16:
It looks like our privacy team is ok with exposing whether an adapter exists. There's some precedent for this sort of question in [`Presentation.getAvailability()`](http://w3c.github.io/presentation-api/#getavailability), and their use of a Promise reminds me that this should probably also be behind a promise, so we don't need to send Bluetooth information to every renderer continuously.

beaufortfrancois, 2015-07-17:
Thanks Jeffrey. Is this what expect from `navigator.bluetooth.getAvailability`?
 
```js  
navigator.bluetooth.getAvailability()
.then(availability => {
  // availability.value may be kept up-to-date by the UA as long as the availability
  // object is alive. It is advised for web developers to discard the object as soon 
  // as it's not needed.
  handleBluetoothAvailabilityChange(availability.value);
  availability.onchange = () => { handleBluetoothAvailabilityChange(this.value); };
})
.catch(() =>
  // Bluetooth availability monitoring is not supported by the platform.
});
```

jyasskin, 2015-07-21:
Roughly. I'm not sure of the name of the getter and member, and whether we can have people use `Object.observe()` on the member instead of the `onchange` event. I believe we can forbid the platform from rejecting the resulting promise (just return `false` if the user's forbidden access). So we might wind up with:

```javascript
navigator.bluetooth.hasAdapter()
.then(adapterInfo => {
  // adapterInfo.value will be kept up-to-date by the UA as long as the availability
  // object is alive. It is advised for web developers to discard the object as soon 
  // as it's not needed.
  handleBluetoothAvailabilityChange(adapterInfo.value);
  Object.observe(adapterInfo, () => {
    handleBluetoothAvailabilityChange(adapterInfo.value);
  });
});
```

Or

```javascript
navigator.bluetooth.getAvailability().then(availability => use(availability.available))
```

`getAvailability()` extends to a future where we might give explicit access to particular adapters, or tell developers whether BT is disabled, or whatever. `hasAdapter()` is more targeted, which avoids getting hopes up, and lets us optimize more for the specific information the developers are getting.

g-ortuno, 2015-07-30:
If the adapter is turned off, is it OK to reject the promise with:

`NetworkError, "Bluetooth Adapter is off"`

?

jyasskin, 2015-07-30:
Per https://github.com/WebBluetoothCG/web-bluetooth/issues/127#issuecomment-119698067, I think we should [reject with a `NotFoundError`](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetooth-requestdevice). In the long run, that'll happen naturally because they cancelled the dialog after getting the "turn on your adapter" error. `NetworkError` is also plausible, but I weakly prefer `NotFoundError`.

g-ortuno, 2015-07-30:
Sounds good. Opened http://crbug.com/515573

commit 0f859805 (2016-09-12): Expose whether Bluetooth is available. (#287)

And add a section describing the new privacy risk.

Fixes #127
closed 2016-09-12 by jyasskin