whatwg/bluetooth #130: Describe what to do if a method is called on a removed attribute

jyasskin, 2015-06-10:
Services, Characteristics, and Descriptors can be removed from a device while an execution context has an object referring to them. If a method is then called on that object, we need to specify that an exception is thrown.

If the same attribute is later re-added to the device, we should probably re-vivify its object.

jyasskin, 2015-06-10:
How does [InvalidStateError](https://heycam.github.io/webidl/#invalidstateerror) sound for the "you used a removed attribute" exception? NotFoundError is also plausible, but it could be confused with [`presentService.getCharacteristic("not_found_uuid")`](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothgattservice-getcharacteristic) (which currently returns a [SyntaxError](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothuuid-getcharacteristic) or `null` or the result of #124).

scheib, 2015-06-10:
`InvalidStateError` sounds reasonable to me.

A related Apps will receive an event `serviceremoved` or `servicechanged`, and can then update any references to objects which are no longer valid, and thus could be written to avoid dealing with this exception what so ever. However, should they fail to do so `InvalidStateError` is good.

jyasskin, 2016-08-15:
Fixed by 001c3a8750c0e9782b19cadcf44c44d073dab0d1.

closed 2016-08-15 by jyasskin