whatwg/bluetooth #135: Scan by device name

scheib, 2015-06-25:
Some devices do not advertise Service UUIDs. E.g. the TI Sensor Tag supports many services, but advertises none. For this type of device to be discovered in the current specification the user agent would need to connect to the device and run a Service Discovery Procedure.

However, if requestDevice's options in BluetoothScanFilter include a name field then many of these devices would be discoverable and could be listed in the chooser UI. The list of Service UUIDs would still need to be provided and used to restrict access.

An open question: Should a device matching a name be displayed in the chooser UI only after running the Service Discovery Procedure and verifying that the devices supports the BluetoothScanFilter.services? This would allow websites to cause devices to be aware of scanning, and would consume more power. Or, should the device be listed and selectable by name match only? This may result in user frustration if a device name matches but upon connection fails to support the required services.

pzboyz, 2015-07-02:
Devices which really want to minimise their current consumption will very likely send nothing in their adverts.

So you need to have a scheme which allows for and deals with that.

At the most they might include their device name in the SCAN_RSP packet, if active scanning is enabled.

jyasskin, 2015-07-02:
@pzboyz Can you name some concrete devices that do this, and describe how their native apps figure out which device to connect to?

The current spec matches iOS's preference: [the CBCentralManager provides only a way to filter by service UUID.](https://developer.apple.com/library/ios/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/PerformingCommonCentralRoleTasks/PerformingCommonCentralRoleTasks.html#//apple_ref/doc/uid/TP40013257-CH3-SW3)

Android's full set of scan filters is at http://developer.android.com/reference/android/bluetooth/le/ScanFilter.Builder.html. I'm not sure which set of those we'll eventually want to support, or exactly how they'll affect the security model.

A filter by device name, especially, seems like it'll encourage a really confusing chooser dialog: "Which of these 5 identical 'Charge HR' devices is yours?" (This already happens, and is enabled by Android's filters.) We may need to put a distance indicator in there after all.

scheib, 2015-07-06:
The CBCentralManager limitation will be frustrating.

The last point, "A filter by device name ..." will be true also if we only scan for devices supporting specific services, if 5 identical Chrage HR devices are in proximity then scanning by name or services will result in the same list.

beaufortfrancois, 2015-08-24:
For info, the Parrot Rolling Spider drone doesn't send any advertisement data. 
Device name and manufacturer data are the only clues to detect that this indeed is the drone.

jyasskin, 2015-08-25:
Per https://github.com/voodootikigod/node-rolling-spider/blob/8996973eaee36be829d984d4ee7e37b0152b379d/lib/drone.js#L93, the Rolling Spider needs a prefix match on the name.

So: 
```javascript
  filters: [
    {name: 'ExactName1'},
    {name: 'ExactName2'},
    {namePrefix: 'Prefix1'},
    {namePrefix: 'Prefix2'}
  ]
```

beaufortfrancois, 2015-10-07:
Can we prioritize this issue?

commit fe26a18d (2015-11-03): Allow filtering by device name and name prefix in requestDevice().

Fixes #135.
closed 2015-11-03 by jyasskin