whatwg/bluetooth #137: Provide a way to pair when the user's paying attention

jyasskin, 2015-07-07:
https://webbluetoothcg.github.io/web-bluetooth/#error-handling says that when the UA receives an "Insufficient Authentication" error, it should "attempt to increase the security level of the connection". However, increasing the security level may require pairing, which may require the user to interact with both their OS and the device, which we should only ask them to do after the site tells them what's going on.

We should 1) give the site a specific error message saying that it needs to ask the user to pair the device, and 2) give the site a way to trigger pairing when the user says they're ready. Triggering pairing should probably use the "[allowed to show a popup](https://html.spec.whatwg.org/multipage/browsers.html#allowed-to-show-a-popup)" language that `requestDevice()` uses.

beaufortfrancois, 2016-01-06:
From what I understood, MacOS will automatically trigger a pairing process when attempting to read/write a "protected" characteristic. @jeromelebel can you confirm?

jyasskin, 2016-09-08:
Moving to a later milestone because we haven't seen folks trying to use the API with devices that need pairing.

jyasskin, 2017-01-28:
#352 gives the UA the option of bonding while connecting. It seems like that's the best we can do, given the absence of a "request bonding" function on MacOS.

jracle, 2017-01-31:
(Cross-referencing #346)

> Moving to a later milestone because we haven't seen folks trying to use the API with devices that need pairing.

Sorry @jyasskin for that, at Logitech we do!! 

Our BLE devices, talking there about existing mice and keyboards, implement HID over GATT service, that requires pairing. They also expose proprietary characteristic that does not always require bonding.

I successfully used web-bluetooth on Mac OS and Chrome OS (with few issues due to the stack or pending bugs). I opened an issue on Android [(670281)](https://bugs.chromium.org/p/chromium/issues/detail?id=670281), that led to a discussion with @beaufortfrancois while you were OOO.

Of course  my experience on Mac OS is also that here bonding is implicit, and is triggered by system upon insufficient privilege error.

g-ortuno, 2017-01-31:
@jracle: You mentioned that macOS works fine. Is it because macOS pairs as soon as it sees the HID services or does the website write to any characteristics that require pairing for which macOS initiates the procedure?

beaufortfrancois, 2017-02-01:
Here's what I get on macOS when connecting my MX Master mouse from https://googlechrome.github.io/samples/web-bluetooth/device-disconnect.html?namePrefix=MX:

```
...
Feb 01 10:44:09.240	ATT Send	FC:E9:FE:31:4E:62	0x0040	Read Request - Handle:0x0021
Feb 01 10:44:09.740	ATT Receive	FC:E9:FE:31:4E:62	0x0040	Error Response - Attribute Handle: 0x0021 - Error Code: 0x05
Feb 01 10:44:09.741	SMP Send	FC:E9:FE:31:4E:62	0x0040	LE SMP Pairing Request Command 
Feb 01 10:44:09.763	SMP Receive	FC:E9:FE:31:4E:62	0x0040	LE SMP Pairing Response Command 
Feb 01 10:44:09.764	SMP Send	FC:E9:FE:31:4E:62	0x0040	LE SMP Pairing Confirm Command 
Feb 01 10:44:09.785	SMP Receive	FC:E9:FE:31:4E:62	0x0040	LE SMP Pairing Confirm Command 
Feb 01 10:44:09.937	SMP Send	FC:E9:FE:31:4E:62	0x0040	LE SMP Pairing Random Command 
Feb 01 10:44:10.314	SMP Receive	FC:E9:FE:31:4E:62	0x0040	LE SMP Pairing Random Command 
Feb 01 10:44:10.438	HCI COMMAND	00:00:00:00:00:00	0x401C	[2019] LE Start Encryption - Connection Handle: 0x0040
Feb 01 10:44:10.439	HCI EVENT	00:00:00:00:00:00	0x0000	Command Status - LE Start Encryption
...
```

For info, full log are available at http://pastebin.com/raw/XS3xVMfM

The Error Code 0x05 is `BLE_HCI_AUTHENTICATION_FAILURE` which indicates that when macOS tries to discover all services/characteristics/descriptors, it stumbles upon one that requires authentication and simply triggers the SMP Pairing Request.

jracle, 2017-02-01:
Thanks a lot for your findings @beaufortfrancois , you were quicker than me to try (I'm waiting for my new macbook pro.. with touchbar.. and enough disk space.. to come, hopefully I'll have enough space to debug)..

Indeed it seems that [CBCentralManager connect:...] attempts reading HID characteritics.

One thing in our case is that we can't act upon HID service, being black-listed by web-bluetooth.. I don't know if in that case we could make an exception on other platforms, attempting pairing if that service is encountered. That would make consistency across platforms, while still keeping the option to pair other services.

beaufortfrancois, 2017-02-08:
@jyasskin @scheib For the sake of consistency across OSes, shall we read HID characteristics when connecting to a device that contains a HID service?

Emill, 2017-02-08:
I agree with @scheib that the goal of Web Bluetooth should explicitly _not_ be to pair a HID mouse/keyboard with the system, but rather allow websites to access common peripherals. https://webbluetoothcg.github.io/web-bluetooth/#security-and-privacy even blacklists a website to access HID characteristics. Note that anyone can create a peripheral that exposes both a HID service and a custom service, that is constructed so that what you write to the custom service is immediately echoed back on the HID service, i.e. you could control the user's keyboard or mouse over the web using web bluetooth if the user is near a device you have placed there, as long as the user is tricked into accepting the BLE connection. Such attacks can easily be done at public areas.

beaufortfrancois, 2017-02-08:
I don't think* Apple folks are doing this unintentionally. According to the [WWDC 2013 - Session 703](https://developer.apple.com/videos/play/wwdc2013/703/), it looks like the HID Over GATT Profile is special and built into the system so that it works just like with Bluetooth Classic devices.

> So, you will be able to discover Bluetooth flow energy HID devices directly from the Bluetooth settings pane like a Bluetooth Classic would be.
> And once you are paired with Bluetooth Energy HID Device, iOS will automatically manage the connection and reconnect to it if the connection goes away therefore giving this illusion that it works just like a normal Bluetooth device that customers are used to.

---

Moreover, Chrome OS already pairs BLE HID devices when connected from Web Bluetooth. We already submitted a [patch](https://codereview.chromium.org/2357083002) to improve that experience.

Here's the current state. I'm sure you'll undestand why I'd love consistency there.

Feature/Platform          | Chrome OS | Android | Mac | Linux  |
------------------------- | :-------: | :-----: | :-: | :-----: |
Pair HID automatically |   ✓   |          |  ✓     |  ✓                |

jracle, 2017-02-09:
Thanks guys for your inputs!

@Emill I get your points concerning security hole we introduce here (mentionned [here](https://bugs.chromium.org/p/chromium/issues/detail?id=670281#c25) as well for the other readers). You also state in issue #346 that device could issue a pairing request to force host present system pairing dialog. That is also something to consider.

Now as @beaufortfrancois says, we also look for consistency across OSes. Also I think Apple just implemented what was defined in HOGP spec (done with Logitech back in 2011).

Also that would really be weird to be able to connect to our HID devices and configure them, while not being able to use them further. It breaks the whole physical web UX, and makes web-bluetooth use much less attractive for us.