whatwg/bluetooth #138: Specify what happens when a device is paired, which reveals its identity address.

jyasskin, 2015-07-15:
A UA may discover a device that's using a private address, and expose that device to a web page with a particular [`BluetoothDevice.instanceID`](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothdevice-instanceid). If the device is later paired, the UA learns the device's identity address and IRK. The spec needs to be clear that the device needs to keep the original `instanceID` through the pairing process.

If the device is discovered again using a resolvable private address, the UA can match it to its public address using the IRK, and can use the same `instanceID`. No problem here.

If the device is discovered again using a non-resolvable private address, it'll get a new `instanceID`. If it's later paired again, and it winds up having the same identity address as a previously-paired device, we can't magically cause it and the previous device to have the same `instanceID`. Should we add a field to devices to point out their canonical instance? An event that fires when this field gets populated? Just leave them as two distinct devices and hope people don't pair devices with non-resolvable addresses much?

jyasskin, 2016-08-30:
I'm leaning toward not telling a site that a [`BluetoothDevice.id`](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothdevice-id) it was previously paired with turns out to be the same physical device as one it was more recently paired with. A subsequent connection to that device should use the more recent ID.

reillyeon, 2019-09-16:
Discussed at 2019 TPAC F2F. Closing this issue as outdated. The `instanceId` field is no longer part of the specification. The `id` field will be the same for a device once it has been paired but can be implemented by the user agent in a way that prevents it from being used to identify the device across multiple origins.

closed 2019-09-16 by reillyeon