whatwg/bluetooth #139: Fix "ID" capitalization

jyasskin, 2015-07-16:
In https://github.com/w3c/web-nfc/issues/27#issuecomment-122013007 @domenic says "Id" should only have an initial capital. @annevk points out in https://github.com/jyasskin/web-bluetooth-1/commit/8154c44f9857336bb5f1d93a0aadfa3a1c19dc12#commitcomment-12208486 that I changed it the other way. https://github.com/w3ctag/design-principles/issues/15 doesn't have explicit guidance for this yet. We could avoid this by just giving `BluetoothDevice`s "id"s instead of "instanceId/D"s.

domenic, 2015-07-16:
> We could avoid this by just giving BluetoothDevices "id"s instead of "instanceId/D"s.

Unless "instance id" is a term of art, this sounds like a nice simplification :)

Otherwise, I can try to do some more serious research to make sure I am correct about the platform preferring `Id` to `ID`.

closed 2015-07-28 by jyasskin

g-ortuno, 2015-10-27:
what about vendorIDSource, vendorID and productID?

domenic, 2015-11-11:
Results:

- NonElementParentNode's getElementById
- TextTrackList's getTrackById
- TextTrackCueList's getCueById
- VTTRegionList's getRegionById
- RTCPeerConnection's getStreamById
- HTMLMediaElementAudioOutputDevice's setSinkId
- DocumentType's publicId and systemId
- MediaKeyEvent's sessionId
- PointerEvent's pointerId
- MessageEvent's lastEventId

There are a few capital "ID"s, on navigator in particular. But the trend is clear. I will update https://w3ctag.github.io/design-principles/#casing-rules to call out "Id" explicitly.

PR #147 (2015-07-28): Rename BluetoothDevice.instanceID to BluetoothDevice.id.
This avoids questions about how to capitalize "ID". There aren't any other uses
of instanceID left, since #113 removed them, so we don't need this spelling to
be consistent.

Fixes #139.

Demo at https://rawgit.com/jyasskin/web-bluetooth-1/rename-instanceid/index.html.
commit 7857fa11 (2015-07-28): Rename BluetoothDevice.instanceID to BluetoothDevice.id.

This avoids questions about how to capitalize "ID". There aren't any other uses
of instanceID left, since #113 removed them, so we don't need this spelling to
be consistent.

Fixes #139.