whatwg/bluetooth #141: Discover "non-discoverable" beacons

beaufortfrancois, 2015-07-18:
There should be a way to discover becaons that have just the "BR/EDR Not Supported" flag set, not any of the "discoverable" flags. An option in `navigator.requestDevice` could look like this.

```js
var uriBeaconUUID = BluetoothUUID.getService(0xFED8);
navigator.bluetooth.requestDevice({
  filters: [{ services: [uriBeaconUUID] }],
  mode: 'findEverythingWeCan' 
}).then(...);
```

Background:
_ChromeOS assumes the point of discovering devices is to connect to them, and a non-discoverable device is probably not connectable either, and because BLE provides the "General Discovery Procedure" to search for devices, and that says "If the Flags AD type (see [Core Specification Supplement], Part A, Section 1.3) is present and either the LE General Discoverable Mode flag is set to one or the LE Limited Discoverable Mode flag is set to one then the Host shall consider the device as a discovered device, otherwise the advertising data shall be ignored."_

armansito, 2015-07-18:
I'm confused about this issue. Why do you want to discover things that are not discoverable? There is a reason for being non-discoverable and you have already cited the spec saying the host shall ignore devices that aren't discoverable. Also check Core Spec v4.2, Vol 3, Part C "Generic Access Profile", Section 9.2.2 "Non-Discoverable" Mode, that reiterates what you cited: "A device configured in non-discoverable mode will not be discovered by any device that is performing either the general discovery procedure or the limited discovery procedure."

So this is a clear WontFix to me.

Also the discovery flags have nothing to do with connectability; if you receive ADV_NONCONN_IND then the device isn't connectable. A device can be non-discoverable but connectable, which is a common case for ADV_DIRECT_IND.

jyasskin, 2015-07-18:
My main goal here is to make sure implementations are interoperable. Nobody's exposing a knob to configure which sorts of flags are returned by a scan, so we'll need to do a survey of existing platforms to see what they do, and see which can be fixed. "Fortunately", it seems like the existing Physical Web beacons (@scottjenson, @matthewsibigtroth) advertise without any of the discoverable flags set, and expect to be discovered, so we can write simple apps on each platform to check whether they're discovered.

Also, naïve question: What's the point of non-discoverable mode with non-connectable advertising events? If we're not supposed to discover the device to process the AD contents, and we're not supposed to connect to a known device, what can we do with it? 3.C.9.2.2 explicitly anticipates this state, and seems to say that the Peripheral can still be connected to?

beaufortfrancois, 2015-07-20:
For info, android [`onLeScan`](http://developer.android.com/reference/android/bluetooth/BluetoothAdapter.LeScanCallback.html#onLeScan(android.bluetooth.BluetoothDevice, int, byte[])) returns all bluetooth LE devices. See Physical Web Client source code at https://github.com/google/physical-web/blob/bcd133ed2ee74a33368839cd237b9ef2b3c89c1d/android/PhysicalWeb/app/src/main/java/org/physical_web/physicalweb/BlePwoDiscoverer.java#L61

g-ortuno, 2015-07-20:
Same in iOS. The Physical Web iOS client returns all devices regardless of the flag.

scottjenson, 2015-07-20:
This is a big confusion in the spec. Many are interpreting the Discoverable bit as "GATT services are discoverable" This is most easily seen in the Nordic BLE scanner app which uses the Discoverable bit as a indicator to show the "Connect" button or not. If *Nordic* is doing this, it's a strong sign that this part of the spec is not agreed upon by the community.

My point is simple: this is not clear cut and to blindly adhere to the spec when very few are following it seems inadvisable at this point.

jyasskin, 2015-07-20:
APIs that scan for devices, like Android's and iOS's, may just be doing a non-standardized procedure, rather than than the standardized General Discovery procedure. Right now, the WebBT spec requires `requestDevice()` to do the standardized procedure, but it'd be straightforward to specify something else that ignores the Discoverable bits. I haven't yet figured out which behavior I prefer.

Nordic may just be working around a hole in the Android API: Android [provides the flags](https://developer.android.com/reference/android/bluetooth/le/ScanRecord.html#getAdvertiseFlags()), but not whether the AD packet was connectable. iOS, on the other hand, [provides whether the packet was connectable](https://developer.apple.com/library/ios/documentation/CoreBluetooth/Reference/CBCentralManagerDelegate_Protocol/index.html#//apple_ref/doc/c_ref/CBAdvertisementDataIsConnectable), but not the flags.

jyasskin, 2015-07-20:
How much of a problem would it be to say that in v1 of this API, `requestDevice()` can't find "pure" beacons that don't set a Discoverable bit. We'd explicitly filter on that in Android, and we'd have to leave enough room in the spec to let us filter on connectability in iOS. We already know that we need to add better support for scanning in a future version, but the current API is really about finding particular devices so a site can communicate with them, and beacons that have opted out of discoverability may not fit that.

beaufortfrancois, 2015-07-21:
For reference, here's a btmon trace of `navigator.bluetooth.requestDevice`:

```js
Bluetooth monitor ver 5.30
= New Index: A4:17:31:78:A6:04 (BR/EDR,USB,hci0)                [hci0] 0.629490
< HCI Command: LE Set Random Address (0x08|0x0005) plen 6       [hci0] 3.946329
        Address: 1B:B0:17:62:80:06 (Non-Resolvable)
> HCI Event: Command Complete (0x0e) plen 4                     [hci0] 3.947497
      LE Set Random Address (0x08|0x0005) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Parameters (0x08|0x000b) plen 7      [hci0] 3.947528
        Type: Active (0x01)
        Interval: 11.250 msec (0x0012)
        Window: 11.250 msec (0x0012)
        Own address type: Random (0x01)
        Filter policy: Accept all advertisement (0x00)
> HCI Event: Command Complete (0x0e) plen 4                     [hci0] 3.948496
      LE Set Scan Parameters (0x08|0x000b) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2          [hci0] 3.948525
        Scanning: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                     [hci0] 3.949496
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
@ Discovering: 0x01 (7)
> HCI Event: LE Meta Event (0x3e) plen 33                       [hci0] 4.221492
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Non connectable undirected - ADV_NONCONN_IND (0x03)
        Address type: Random (0x01)
        Address: C5:34:F9:7A:00:40 (Static)
        Data length: 21
        Flags: 0x04
          BR/EDR Not Supported
        16-bit Service UUIDs (complete): 1 entry
          Unknown (0xfed8)
        Service Data (UUID 0xfed8): 00f200676f6f676c6507
        RSSI: -71 dBm (0xb9)
@ Device Found: C5:34:F9:7A:00:40 (2) rssi -71 flags 0x0004
        02 01 04 03 03 d8 fe 0d 16 d8 fe 00 f2 00 67 6f  ..............go
        6f 67 6c 65 07                                   ogle.           
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2          [hci0] 4.421283
        Scanning: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                     [hci0] 4.422591
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2          [hci0] 4.422653
        Scanning: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                     [hci0] 4.423569
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 33                       [hci0] 5.222490
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Non connectable undirected - ADV_NONCONN_IND (0x03)
        Address type: Random (0x01)
        Address: C5:34:F9:7A:00:40 (Static)
        Data length: 21
        Flags: 0x04
          BR/EDR Not Supported
        16-bit Service UUIDs (complete): 1 entry
          Unknown (0xfed8)
        Service Data (UUID 0xfed8): 00f200676f6f676c6507
        RSSI: -78 dBm (0xb2)
@ Device Found: C5:34:F9:7A:00:40 (2) rssi -78 flags 0x0004
        02 01 04 03 03 d8 fe 0d 16 d8 fe 00 f2 00 67 6f  ..............go
        6f 67 6c 65 07                                   ogle.           
[... 78 lines cut ...]
          Unknown (0xfed8)
        Service Data (UUID 0xfed8): 00f200676f6f676c6507
        RSSI: -74 dBm (0xb6)
@ Device Found: C5:34:F9:7A:00:40 (2) rssi -74 flags 0x0004
        02 01 04 03 03 d8 fe 0d 16 d8 fe 00 f2 00 67 6f  ..............go
        6f 67 6c 65 07                                   ogle.           
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2          [hci0] 8.439321
        Scanning: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                     [hci0] 8.440568
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2          [hci0] 8.440664
        Scanning: Enabled (0x01)
        Filter duplicates: Enabled (0x01)
> HCI Event: Command Complete (0x0e) plen 4                     [hci0] 8.441544
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Set Scan Enable (0x08|0x000c) plen 2          [hci0] 8.830942
        Scanning: Disabled (0x00)
        Filter duplicates: Disabled (0x00)
> HCI Event: Command Complete (0x0e) plen 4                     [hci0] 8.831467
      LE Set Scan Enable (0x08|0x000c) ncmd 1
        Status: Success (0x00)
@ Discovering: 0x00 (7)

beaufortfrancois, 2015-07-22:
I've just tried with an Eddystone beacon and while it sets the "LE General Discoverable Mode" flag, it still doesn't show up in the bluetooth console with `scan on`.

```js
> HCI Event: LE Meta Event (0x3e) plen 43
      LE Advertising Report (0x02)
        Num reports: 1
        Event type: Connectable undirected - ADV_IND (0x00)
        Address type: Random (0x01)
        Address: E8:FC:B7:D1:91:B9 (Static)
        Data length: 31
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        16-bit Service UUIDs (partial): 1 entry
          Tx Power (0x1804)
        Company: StickNFind (249)
          Data: 018b5cd5b050e3cb6113c0680003b213c31644f5
        RSSI: -60 dBm (0xc4)

jyasskin, 2015-08-19:
I believe the wording change we need here is just to say that it's conforming to fail to check for a Discoverable flag during the General Discovery procedure. Most platforms will check for it, but Mac and iOS may not.

beaufortfrancois, 2016-06-28:
For info, Bluez still hides non-discoverable BLE devices. See https://kernel.googlesource.com/pub/scm/bluetooth/bluez/+/a74268601eb404447cc3c4a5f0b1f5510c4f2b20%5E%21/#F0

beaufortfrancois, 2016-06-30:
Just filed https://bugs.chromium.org/p/chromium/issues/detail?id=624733 to keep track of this.

jyasskin, 2016-07-25:
@jhedberg pointed out on #bluez-users that the Discoverable flag seems to be used when a peripheral wants to create a long-term relationship with a central, while it tends to be un-set when a peripheral or broadcaster just wants to send information. That implies that `requestDevice()` should check for the Discoverable flag, but `requestLEScan()` shouldn't.

We can't implement that distinction on any platform except BlueZ as far as I can see, and we probably want the smaller BlueZ platform to match the larger Android and iOS platforms for compatibility. So the spec may wind up having a Note about the ideal situation while specifying something less ideal.

PR #267 (2016-08-05): Allow requestDevice() to include non-discoverable devices.
The CoreBluetooth API doesn't allow the UA to check the discoverable flags, so
we can't require it in the spec.

Fixes #141.

Preview at https://api.csswg.org/bikeshed/?url=https://github.com/jyasskin/web-bluetooth-1/raw/discover-non-discoverable/index.bs#scan-for-devices
commit 7a8f851c (2016-08-05): Allow requestDevice() to include non-discoverable devices. (#267)

The CoreBluetooth API doesn't allow the UA to check the discoverable flags, so
we can't require it in the spec.

Fixes #141.
closed 2016-08-05 by jyasskin