whatwg/bluetooth #145: Code example malformed.

carybran, 2015-07-24:
Hi - tested this with U2F work we are doing and was able to successfully see the authenticator!

Found issue in documentation with the request device call:

navigator.bluetooth.requestDevice({
  filters: [{
    services: [0x1234, 0x12345678, '99999999-0000-1000-8000-00805f9b34fb'],
  }]
.then(device => {...})
.catch(error => { console.log(error); });

Should be:

navigator.bluetooth.requestDevice({
  filters: [{
    services: [0x1234, 0x12345678, '99999999-0000-1000-8000-00805f9b34fb'],
  }]})  <===== ADDED closing braces
.then(device => {...})
.catch(error => { console.log(error); });

beaufortfrancois, 2015-07-24:
Will fix it now!
Thank you Cary.

beaufortfrancois, 2015-07-24:
Quick question. 
Do you use HID GATT Service for U2F?

carybran, 2015-07-24:
> <reply+000dc943020edd3790afae8ed7e3011139d1b1ebfe17b6b692cf0000000111ca5f839
> 2a169ce05ca31fb@reply.github.com>

We are using the FIDO U2F BLE spec -
https://fidoalliance.org/specs/fido-u2f-v1.0-nfc-bt-amendment-20150514/fido-
u2f-bt-protocol.html

<https://github.com/WebBluetoothCG/web-bluetooth/issues/145#issuecomment-124
707891> .

beaufortfrancois, 2015-07-24:
Unless we're talking about a different U2F Service, you may want to talk to @jyasskin then. From what I can read at https://plus.google.com/u/0/+FrancoisBeaufort/posts/bYYfNLQyMtX

_We're going to blacklist the U2F service UUID because direct Bluetooth communication from arbitrary websites breaks their security model. They assume they can trust the Bluetooth message about the origin it's coming from, but a site sending a raw GATT message to a U2F device could spoof that, and the phishing flow becomes too easy.﻿_

carybran, 2015-07-24:
Thank you - That is what I heard too - I am just testing with an early version of the U2F GATT service to see if I could actually communicate with the device using the new API.

beaufortfrancois, 2015-07-24:
We're good then ;)

carybran, 2015-07-24:
Yes!

closed 2015-07-27 by jyasskin