whatwg/bluetooth #151: Support for Android Web browsers

cuttingedge03, 2015-08-18:
Does this library support android Web Browser?

beaufortfrancois, 2015-08-18:
It is supported by Chrome 45 in Chrome OS. Support for Mac, Android, Windows and Linux will follow. Read more about this at https://developers.google.com/web/updates/2015/07/interact-with-ble-devices-on-the-web

cuttingedge03, 2015-08-18:
@beaufortfrancois Is there any tentative date for the subsequent releases?

cuttingedge03, 2015-08-18:
I am working on Physical Web and Web Bluetooth becomes important for dynamic web pages.

beaufortfrancois, 2015-08-18:
I'm as excited as you on Physical Web & Web Bluetooth. We're still in the process of figuring out how to make them work nicely together. See https://lists.w3.org/Archives/Public/public-web-bluetooth/2015Aug/0000.html

There is no ETA for Android. 
Be sure we'll let developers know when this is out ;)

closed 2015-08-18 by cuttingedge03