whatwg/bluetooth #152: Should connectGATT() time out or wait until a device is available?

jyasskin, 2015-08-20:
And if it waits, should that wait be cancellable?

On at least [Android](http://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#connect()) and [iOS](https://developer.apple.com/library/ios/documentation/CoreBluetooth/Reference/CBCentralManager_Class/index.html#//apple_ref/occ/instm/CBCentralManager/connectPeripheral:options:), if you call "connect" on a device, and the device isn't nearby, the call just waits until the device shows up. Bluetooth specifies T<sub>GAP</sub>(adv_slow_interval) <= 1.2s as the longest the central's supposed to have to wait to connect to a present device, so either we could time out the connects, or users could. If we make users do it, it naturally solves #31, but we probably want a way to cancel the attempt to connect. Right now, unlike in the Android and iOS APIs, we only expose [`disconnect()` on the `BluetoothGATTRemoteServer`](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothgattremoteserver-disconnect), which is only available once [`connectGATT()`](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothdevice-connectgatt)'s promise fulfills.

Right now, I'm inclined to say that `connectGATT()` should hang until a device shows up, mention in a Note that 1.2s makes a good timeout, and give some way to cancel the connection, maybe by making `connectGATT()` fill in `BluetoothDevice.gattServer` synchronously, and having folks call `disconnect()` on that.

g-ortuno, 2015-08-21:
so:
```javascript
navigator.bluetooth.requestDevice({...})
  .then(device => {
    device.connectGATT(handle_connect);
    // device.gattServer.connected === false
  });
```
SGTM. One question: Would `gattServer.disconnect()` make the promise reject?

jyasskin, 2015-08-21:
I believe so. I'm open to other opinions.

g-ortuno, 2015-08-21:
Sounds good. It could reject with an [AbortError](https://heycam.github.io/webidl/#aborterror)

scheib, 2015-08-24:
I agree with having the ability to abort. For some devices an unwanted connection may be frustrating (e.g. block other connections). 

Having `connectGatt / disconnect` on different levels of the object hierarchy seems slightly sad, but OK. An alternative would be `device.connectGatt / device.disconnectGatt`.

yrliou, 2015-11-12:
I would also prefer to have the ability to abort, if we decide to revise ```device.connectGatt``` as ```device.gattServer.connect``` as mentioned in #181, we could use ```device.gattServer.disconnect``` to cancel ongoing connection requests.

beaufortfrancois, 2016-06-16:
Following https://bugs.chromium.org/p/chromium/issues/detail?id=620405 discussion, it looks like `device.gatt.connect` will never reject... I'm not a big fan of this. I'd rather have `device.gatt.connect` to fail with the same amount of time on each platform and have developer to reconnect automatically or ask user to reconnect.

```js
function connect() {
  return device.gatt.connect()
  .then(server => ...)
  .catch(e => {
    if (e.name == 'TimeoutError') {
      // e.message = Device failed to connect after X seconds...
      // Automatically try to reconnect
      return connect();
    }
  });
}

beaufortfrancois, 2016-06-16:
For info, Chrome Dev for Android currently returns a "NetworkError: Connection failed for unknown reason" after 30 seconds when trying to connect to a BLE device with the Web Bluetooth API.

jyasskin, 2016-06-16:
I don't like the loop because it keeps the CPU active every X seconds, where a hanging connect could let the CPU sleep and rely on the more power-efficient Bluetooth chip. That said, the loop is plausible if other folks like it more.

beaufortfrancois, 2016-09-14:
- Linux Kernel < 3.15 has a 40s timeout
- Linux Kernel >= 3.15 has a 20s timeout
- Android timeout is 30s
- Mac doesn't have a timeout

I wonder if we should internally call `disconnect()` after a certain amount of time to consistency across platforms. If I'm developing on Android, I might not implement the timeout after x seconds because `device.gatt.connect` would fail after 30s which I'm fine with. However, on Mac, it won't and this will leave my application in a weird state.

I suggest we always call disconnect after 20s to be consistent.

@jyasskin WDYT?

Please close if it doesn't make sense.

jyasskin, 2016-09-15:
https://bugs.chromium.org/p/chromium/issues/detail?id=620405#c6 mentions that there's a way to get BlueZ to do a hanging connect. We should change the code to use that if possible.

On Android, there's also a way to do a hanging connect ([BluetoothGatt.connect()](https://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#connect()) or [BluetoothDevice.connectGatt(..., true, ...)](https://developer.android.com/reference/android/bluetooth/BluetoothDevice.html#connectGatt(android.content.Context, boolean, android.bluetooth.BluetoothGattCallback))), but they may not work until we've connected once to the device with the timeout. We should experiment to see what's possible and then make the spec match.

Emill, 2016-09-15:
How do you mean that connectGatt with autoConnect may not work until we've connected once to the device?
Are your referring to https://code.google.com/p/android/issues/detail?id=69834 ?

g-ortuno, 2016-09-15:
From experimentation we've found that when calling connectGatt with autoConnect = true will never end up connecting to the device so we always try to connect with autoConnect = false [[1]](https://cs.chromium.org/chromium/src/device/bluetooth/android/java/src/org/chromium/device/bluetooth/ChromeBluetoothDevice.java?q=ChromeBluetooth&sq=package:chromium&dr=CSs&l=109)

Emill, 2016-09-16:
@g-ortuno that seems very strange. I've been using autoConnect for years on many different Android devices and never noticed that it fails to connect. Maybe you simply have a peripheral that advertises with a quite big advertisement interval (in the order of seconds rather than 10s or 100s of milliseconds)?

scheib, 2016-09-16:
Android team has recommended to do initial connections without autoconnect, and only re-connections with autoconnect=true. They haven't explained the reasoning behind this. In testing on a nexus 6 I did see  difficulties with autoconnect=true.

We could, as an implementation detail, have Android connect, timeout, and then roll over into an autoconnect.

Emill, 2016-11-29:
I assume the recommendation "use direct connect for first connection, and re-connections with auto" is simply because that the default scan parameters are more active for direct connect, and it has a built-in timeout, which are usually two nice things if you want to have a simple UI to scan and connect a BLE device.

Regarding having multiple direct connections and aborting connections, you should be aware of this bug on Android:
https://code.google.com/p/android/issues/detail?id=228633

Anyway, I find it very annoying the Android team has decided not to use the white list for "direct connect" because a direct connect blocks away all other pending connections.

For the actual initial question, I think connectGATT should simply wait until the device has connected (no timeout). If someone wants a timeout, it's easy to set up one using setTimeout in javascript and disconnect upon timeout. Thanks to javascript's single threading you can be sure you won't get any spurious connected event right after a disconnect command (which is a common race condition in other multi-threaded languages).

NAllred91, 2017-05-07:
In its current state, should `disconnect()` cause `device.gatt.connect()` to reject?  I'm trying to work through some connections issues I encounter when the user refreshes the page and immediately tries to reconnect.

For me, `device.gatt.connect()` hangs and `disconnect()` isn't causing the promise to reject.

scheib, 2017-05-10:
Chrome's implementation does not yet cancel pending connect calls when
disconnect is called.
Star https://bugs.chromium.org/p/chromium/issues/detail?id=684073 for
updates.

wesleyyee, 2017-09-21:
Have there been any updates to this? The connect never resolves if the device does not come back within a minute or so. Thus leaving our app unable to pair to any device at all (via connect or requestDevice)

scheib, 2017-10-03:
There hasn't been an update, thank you for calling out attention to it.  The spec doesn't currently inticate if connect should time out.  But it does require disconnect to terminate pending connections, and that should work for your situation bue doesn't due to an implementation issue https://crbug.com/684073. I can raise the priority of fixing that issue, and if that's sufficient for you I think we should continue disconnect discussion there.

wesleyyee, 2017-10-03:
@scheib thanks! It would be very helpful if that issue were resolved!

wesleyyee, 2017-10-17:
@scheib is there a timeline for this issue being addressed?

scheib, 2017-10-18:
@wesleyyee there is a prioritized work queue.  If you'd like to discuss more you can reach me at scheib and @ domain google.com.

russellrain, 2019-02-04:
Anyone have a decent workaround for this?  For now it seems the only way to cancel once you've started gatt.connect() is to refresh the page since disconnect() doesn't actually abort a connect()

curlyz, 2023-04-28:
2023, MacOS, can't find a way to abort a connect :(

mgibbs189, 2025-01-23:
@scheib are you still involved with Chromium?

Being able to properly handle dropouts/reconnects is quite an important part of using web-bluetooth. It'd be fantastic to have this issue resolved once and for all, hopefully you could help!

scheib, 2025-02-01:
@mgibbs189 thanks for discussing on https://crbug.com/40502943, that seems a good forum to get into more details for now.

PR #280 (2016-09-07): Make disconnect() abort connect().
Also add a note that connect() could wait forever if an appropriate
advertisement isn't received.

Fixes #152.

Preview at https://api.csswg.org/bikeshed/?url=https://raw.githubusercontent.com/jyasskin/web-bluetooth-1/hanging-connect/index.bs#dom-bluetoothremotegattserver-connect.
commit 3dbaeadf (2016-09-07): Make disconnect() abort connect(). (#280)

Also add a note that connect() could wait forever if an appropriate
advertisement isn't received.

Also clarify that disconnect() isn't intended to break the connection if
other non-browser systems are using the connection.

Fixes #152.