whatwg/bluetooth #153: Scan by manufacturerData

beaufortfrancois, 2015-08-24:
It would be nice to be able to scan with filters on manufacturer data.

Here's how it would look like for the Rolling Spider Drone:
```js
var rollingSpiderData = new Map([[0x0043, [0xcf, 0x19, 0x00, 0x09, 0x01, 0x00]]]);
return navigator.bluetooth.requestDevice({
  filters: [{ manufacturerData: [rollingSpiderData] }]
}).then(device => { ... })
```
According to https://github.com/voodootikigod/node-rolling-spider/blob/8996973eaee36be829d984d4ee7e37b0152b379d/lib/drone.js#L94

jyasskin, 2015-08-25:
Android allows scanning by a mask of the manufacturer data bits: http://developer.android.com/reference/android/bluetooth/le/ScanFilter.Builder.html#setManufacturerData(int, byte[], byte[]). Do we want that too?

Do we want to also allow matching just a prefix of the data?

We can allow just exact matches for now, and extend the array/buffersource value to an object if/when developers ask for more.

Should manufacturerData be a Map from id to array/buffersource or an object whose keys are the id? e.g.

```javascript
filters: [{ manufacturerData: {0x0043: [0xcf, 0x19, 0x00, 0x09, 0x01, 0x00]} }]
filters: [{ manufacturerData: new Map([[0x0043, [0xcf, 0x19, 0x00, 0x09, 0x01, 0x00]]] }]
filters: [{ manufacturerData: {0x0043: new UInt8Array([0xcf, 0x19, 0x00, 0x09, 0x01, 0x00])} }]
filters: [{
  manufacturerData: {
    0x0043: {
      bits: new UInt8Array([0xcf, 0x19, 0x00, 0x09, 0x01, 0x00]),
      mask: new UInt8Array([0xff, 0xff, 0x00, 0xff, 0xff, 0x00]),
      matchType: 'prefix',
    }
  }]
```

I'm leaning toward using an object, and allowing `(BufferSource or sequence<octet>)`, but not expanding to the mask or prefix matches yet.

beaufortfrancois, 2015-08-26:
A simple `object` would work well indeed for now.

jyasskin, 2016-05-13:
We should also allow filtering by Service Data.

beaufortfrancois, 2016-08-24:
I believe we can close this issue since https://github.com/WebBluetoothCG/web-bluetooth/pull/265 has been merged and handle `manufacturer`.

closed 2016-08-24 by beaufortfrancois