whatwg/bluetooth #154: Scan by Organizationally Unique Identifier (OUI)

beaufortfrancois, 2015-08-25:
It would be nice to be able to scan with filters on OUI.

Here's how it would look like for the Rolling Spider Drone:
```js
return navigator.bluetooth.requestDevice({
  filters: [{ ouiData: [0x880F10] }] // Huami Information Technology Co.,Ltd.
}).then(device => { ... })
```
According to https://standards.ieee.org/develop/regauth/tut/eui.pdf and http://standards-oui.ieee.org/oui.txt

jyasskin, 2015-08-25:
Where's the OUI in Bluetooth data? If it's in some of the advertising data, this is probably fine, but if it's in https://developer.bluetooth.org/gatt/characteristics/Pages/CharacteristicViewer.aspx?u=org.bluetooth.characteristic.system_id.xml, I believe it's too expensive to connect and gather that data while scanning.

beaufortfrancois, 2015-08-26:
According to http://git.kernel.org/cgit/bluetooth/bluez.git/tree/lib/bluetooth.c#n100, OUI is based on Bluetooth Address. For instance,

bluetooth address | OUI | Company Name |
--- | --- | --- |
88:0F:10:9D:EB:42  | 88-0F-10 | Huami Information Technology Co.,Ltd. |

g-ortuno, 2015-08-26:
Hmm that's going to be tricky on Mac since they don't expose the address.

jyasskin, 2015-08-26:
Yeah, and lots of devices use random addresses, without the APIs telling us that the addresses are random. I don't think we can practically filter scans by OUI.

closed 2015-08-26 by jyasskin