whatwg/bluetooth #163: When a device causes a page to open, the page should be able to find the device

jyasskin, 2015-09-13:
For example, an [Eddystone-URL](https://github.com/google/eddystone/tree/master/eddystone-url) beacon can cause the browser to suggest a user visit a particular web page. If the user does, that page could get direct access to the device instead of needing to go through the `requestDevice()` chooser.

Sketch at https://docs.google.com/document/d/1jFCTyq84T2fLc8ZhxorTz3u_gCk59hp17EmkFgaDQ2c/edit?usp=sharing.

mfoltzgoogle, 2015-10-27:
Thanks for the document @jyasskin, that explains pretty well what I was thinking of regarding the question I asked at yesterday's F2F.

FWIW I like the approach using `navigator.bluetooth.referringDevice`, as it makes the page initialization code pretty clean.  Also things appended to the URL might get lost or munged during redirects.

```
if (navigator.bluetooth.referringDevice) {
  connectToServices(navigator.bluetooth.referringDevice);
} else {
  connectDeviceButton.onclick = function() { navigator.bluetooth.requestDevice
      .then(connectToServices); };
}
```

jyasskin, 2015-10-29:
Before we can do this, we need to ensure it doesn't make it really easy for a hostile physical-web device to get paired as a keyboard: https://github.com/google/physical-web/issues/568. Thanks @hillbrad for pointing out the problem during TPAC.

beaufortfrancois, 2016-01-06:
If we go down the Permissions API road (https://lists.w3.org/Archives/Public/public-web-bluetooth/2015Dec/0007.html), it would be as simple as adding a `referringDevice` key to the Permission object.

```js       
permissions.query({ name: 'bluetooth' })
.then(permission => {
  if (permission.state != 'granted') {
    console.log('User has forbidden websites to use bluetooth...');
    return;
  }
  if (permission.referringDevice) {
    // Let's try to connect to the Physical Web object first. 
    return connectDevice(permission.referringDevice);
  }
  ...

PR #230 (2016-07-27): Specify the Eddystone upgrade.
Preview at https://rawgit.com/jyasskin/web-bluetooth-1/eddystone-upgrade/index.html.

Much of the content in this patch is about letting a UA not constrain the set of services a site can access on a device that asked to open it. The threads at https://lists.w3.org/Archives/Public/public-web-bluetooth/2016Apr/0000.html and https://plus.sandbox.google.com/+FrancoisBeaufort/posts/jGBu1kxHbpZ didn't come to any firm conclusions on this point, but I think it makes sense to assume that a site advertised by a device is cooperating with that device, and so the normal pur [... cut]
commit a2f34f65 (2016-07-27): Specify the Eddystone upgrade. (#230)

Fixes #163
closed 2016-07-27 by jyasskin