whatwg/bluetooth #166: Serialize start/stop notifications

jyasskin, 2015-09-22:
Right now, [startNotifications()](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothgattcharacteristic-startnotifications) and [stopNotifications()](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothgattcharacteristic-stopnotifications) run all of their steps in parallel, meaning that

```javascript
characteristic.startNotifications();
characteristic.stopNotifications();
```

can leave the notifications enabled. We should find wording to exclude that race.

scheib, 2015-10-03:
I think we should avoid complexity by:
- Disallowing multiple contexts to access the same device. (issue #169)
- Allowing these operations to operate in parallel. I think an application is broken if making async calls with dependencies races between them, and I don't think we should try to enforce a async queue.

marcoscaceres, 2015-10-07:
I guess one could have a flag that indicate start is running, and hence stop needs to wait until that flag is removed before it runs?

marcoscaceres, 2015-10-07:
Worst case, I guess you could immediately reject `*Notifications()` if either is not in "stable" stated. I don't think that's great tho. As each returns a promise, there would not need to be a race so long as the internal state of start/stop is known.

marcoscaceres, 2015-10-07:
(fixed typo above)