whatwg/bluetooth #168: Consider exposing whether a characteristic has notifications enabled.

jyasskin, 2015-09-24:
Should we expose a boolean on [`BluetoothGATTCharacteristic`](https://webbluetoothcg.github.io/web-bluetooth/#bluetoothgattcharacteristic) that's true while the characteristic is subscribed to GATT notifications? It's convenient for testing the implementation, and might be helpful for other libraries.

beaufortfrancois, 2015-09-28:
I would love this indeed!

VinceGa, 2016-03-23:
If we expose descriptors of a characteristic and let the developers check if CCCD(0x2902) exists, they will check/flip the flag and take care of the rest.

g-ortuno, 2016-03-23:
@VinceGa we currently don't allow developers to flip the descriptor themselves. https://github.com/WebBluetoothCG/registries/blob/master/gatt_blacklist.txt#L29

VinceGa, 2016-03-23:
if we allow them to subscribe/unsubscribe(startNotification/stopNotification), we still let a web page interfere with other pages. 
No? Did I miss anything?
Thanks.

jyasskin, 2016-03-23:
If 2 pages are subscribed to notifications, and one of them calls [`stopNotifications()`](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothremotegattcharacteristic-stopnotifications), that doesn't unsubscribe the other one, in the current specification. It's true that exposing descriptor access would give us a uniform API to read the state, and to control whether notifications or indications get enabled, but I think we'd still want to intercept uses of that descriptor to prevent pages from interfering with each other.

VinceGa, 2016-03-23:
Do the 2 pages on the same device share the same CCCD?
If I add a descriptor for my service, can I write to it in our current spec?

Thanks for your time.

jyasskin, 2016-03-23:
Because there's only 1 physical Bluetooth connection, and only 1 bond, yes the two pages share the same CCC descriptor, unless the browser explicitly intercepts uses and makes it look like they're separate. Marcel Holtmann suggested that Bluetooth Core might eventually provide a way for controllers to pretend to be multiple devices, one per application, where they'd each get their own CCC descriptor, but that'll take a while to propagate through platform APIs.

The current spec does have a whole [`BluetoothRemoteGATTDescriptor`](https://webbluetoothcg.github.io/web-bluetooth/#bluetoothremotegattdescriptor) type for messing with device-defined descriptors. It's not implemented in Chromium yet, but it's in the queue, and could get done faster with concrete use cases.