whatwg/bluetooth #169: Allow device connections from only one script context at a time.

scheib, 2015-10-03:
Connection, notification, read & writes, are all problematic if multiple script contexts can be issuing commands to one device. I propose that the specification only permit connection to a device from one context.

beaufortfrancois, 2015-10-05:
Let's say I open a browser tab and starts to interact with a nearby BT device.
Then, I open a second tab (same URL) and perform "requestDevice". How will user know that selecting the previous device will actually interrupt the first tab there?

scheib, 2015-10-05:
We have a few options:
- Reject second connection with an error that it is in use already. Web app must communicate to user that they need to stop use from other tab.
- Display UI upon connection request prompting user if they want to disconnect other tab or not.
- Just disconnect other app, the forground works. Apps will always be susceptible to disconnections, so this is just one more scenario.

beaufortfrancois, 2015-10-07:
I'd personally vote for option 3 but I'd love to hear what other think.
I may add a specific JS error so that developer could explain what went wrong to user so that he can understand.

jyasskin, 2015-10-12:
I'm really reluctant to add this restriction on developers and users. We should prefer making our implementation somewhat more complex so that websites and user experiences can be less complex.

If we do it, I think "just disconnect the other page" is the only plausible way: we shouldn't show UI on an event the user might not be expecting, and we can't give the web page enough information for it to tell the user which other page to close.

jracle, 2016-07-26:
If I could add a fourth option here : don't disconnect device (aka NOP™)..

At Logitech, our devices use to communicate with multiple apps at the same time, leaving underlying app / protocol responsibility to sort its own requests / responses from the whole.

If firmware and GATT driver supports it, why should we have to limit number of connections to our peripheral?

Furthermore, OS also maintains connections in case your device has Hid over GATT profile. So you might end-up closing connection to those mice / keyboards.

I'll also open another issue concerning behavior of GATT server disconnect.

closed 2016-09-19 by jyasskin