whatwg/bluetooth #170: Object destroyed before request is resolved.

g-ortuno, 2015-10-05:
For `startNotifications()` if the characteristic object is destroyed before step 5 should we still start a gatt procedure to start notifications?

beaufortfrancois, 2015-10-07:
Could you summarize steps 1 to 4 there?

g-ortuno, 2015-10-07:
@beaufortfrancois Done.

jyasskin, 2016-02-02:
Because of the cache in [[[attributeInstanceMap]]](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetooth-attributeinstancemap-slot), characteristics aren't destroyed until the global object goes away, roughly when the tab is navigated or closed. I can see arguments both for and against having the UA avoid subscribing to notifications until there's some javascript code around that could observe the subscription, but because that delay isn't observable except for the delay in a value update, I don't think we need any change to the spec to allow it.

Closing, but feel free to re-open if I've missed something.

closed 2016-02-02 by jyasskin