whatwg/bluetooth #171: Should GetGATTChildren required connected == true?

scheib, 2015-10-05:
getPrimaryService and other methods rely on GetGATTChildren algorithm to obtain GATT objects. Should they fail if not connected?

The Chromium implementation currently attempts to obtain this information even when not connected. It could be cached, so it is possible to allow implementations to attempt to service these requests, but it would be cleaner to specify either that they must attempt cached results or reject with a known error.

jyasskin, 2015-10-14:
Probably yes. This is also related to #114. I think we have several holes around handling disconnected devices.

jyasskin, 2015-10-14:
I think I'd rather fail fast on a disconnected device, probably with a `NetworkError`, rather than allowing cached results.

g-ortuno, 2015-10-15:
This seems like something that should be added to the [Populate Cache Algorithm](https://webbluetoothcg.github.io/web-bluetooth/#populate-the-bluetooth-cache). Something like: If device.connected == false then return a Network Error.

commit 7d590bad (2016-03-09): Enforce that the current realm is connected while running GATT procedures.

This change makes it so that a disconnect() call takes effect immediately and
prevents any subsequent changes to member variables from a returning GATT
procedure, even if the GATT procedure started before the disconnection. It also
forces all active requests to fail with a NetworkError so that no remnants of
the connection keep running after it's terminated.

This reverses the readValue() promise and event order from #175, in order to
avoid needing to perform a microtask checkpoint inside the algorithm.

Fixes #114. Fi [... cut]
closed 2016-03-09 by jyasskin