whatwg/bluetooth #180: Figure out how to limit access from iframes

jyasskin, 2015-10-26:
During the TPAC session, the question came up of whether cross-origin iframes should be able to call `requestDevice` or communicate with devices they already have access to. See @torgo's https://github.com/w3ctag/spec-reviews/issues/90. I believe the current Chromium implementation completely forbids access from iframes, but we should figure out if that's the right thing to specify.

beaufortfrancois, 2016-01-06:
For info, I'm able to request a BT device from an iframe in Chrome OS 49.0.2587.3.

beaufortfrancois, 2016-11-02:
As of today, Web Bluetooth doesn't work on cross-origin iframes but works on same-origin iframes. Tested on `Google Chrome 55.0.2883.17`

zcorpan, 2017-02-23:
This could use the same logic as PaymentRequest, right? (Except maybe for Bluetooth we can start using FeaturePolicy to opt in instead of minting an `allowbluetooth` attribute...)

Flo082002, 2023-04-25:
Disable Bluetooth:
`<iframe src="https://example.net/" height="700" width="1200" allow="bluetooth 'none'" title="Iframe Example"></iframe>`

Enable Bluetooth:
 `<iframe src="https://example.net/" height="700" width="1200" allow="bluetooth" title="Iframe Example"></iframe>`

reillyeon, 2023-04-25:
Oops, we forgot to close this issue when Permissions Policy support was added: https://webbluetoothcg.github.io/web-bluetooth/#permissions-policy

closed 2023-04-25 by reillyeon