whatwg/bluetooth #181: Unconditionally define the gattServer attribute, and move connect() there

jyasskin, 2015-10-27:
Talking with @yrliou and @brucesunmozilla at TPAC, it looks like moving connect into BluetoothGATTRemoteServer makes more sense than the current arrangement.

beaufortfrancois, 2015-10-27:
To be clear, does that mean we would have something like this?

```js
navigator.bluetooth.requestDevice({filters:[{services:['battery_service']}]})
.then(device => device.gattServer.connect())
.then(gattServer => {
  // Is gattServer returned?
  // Are device.gattServer and gattServer the same?
})

jyasskin, 2015-10-27:
Yes, yes, and yes, respectively. We could also provide device.connectGATT() as an alias for device.gattServer.connect(), if folks think that's useful.

beaufortfrancois, 2015-10-27:
I like it without the alias.

yrliou, 2015-10-29:
We proposed this since it looks more symmetric in this way, it's a bit of weird that connect and disconnect functions are at different levels.
Personally I would vote for no connectGatt alias.

commit 3e1d42c9 (2016-01-29): Fix #181: Unconditionally define device.gattServer, and connect from there.
commit dd4dcaf0 (2016-01-29): Merge pull request #205 from jyasskin/gatt-server-connect

Fix #181: Unconditionally define device.gattServer, and connect from there.
closed 2016-01-29 by jyasskin