whatwg/bluetooth #185: Don't read Extended Properties until necessary

scheib, 2015-10-30:
For each characteristic the spec currently checks the Extended Properties bit of the characteristic properties and if set requires the Characteristic Extended Properties descriptor to be read. This will impose a delay before the characteristic can be returned at all, because access to the properties is synchronous. Thus before each characteristic is returned an additional descriptor read would be required when extended properties is set. Unless  there is a need to have this information pre-fetched I suggest Web Bluetooth not include reliableWrite and writableAuxiliaries in the BluetoothCharacteristicProperties interface and instead allow applications to read the descriptor explicitly as needed.

jyasskin, 2015-10-30:
This was discussed in #36.

beaufortfrancois, 2016-06-08:
What does that mean for this issue since #36 is closed?

jyasskin, 2016-06-08:
Just that there's prior context to be aware of. I don't know what the right answer is.

beaufortfrancois, 2016-06-08:
Should we implement reading Extended Properties on Android for now to bring consistency for the first version of the Web Bluetooth API then?
https://bugs.chromium.org/p/chromium/issues/detail?id=548449

g-ortuno, 2016-06-09:
I don't think resolving the spec or the implementation is as high priority as other issues

beaufortfrancois, 2016-06-10:
I agree. I wanted to see if other contributors would raise their voice and help us make the right decision. Let's wait then.