whatwg/bluetooth #191: Let websites scan for nearby advertisements.

jyasskin, 2015-12-09:
Websites should be able to register for an event when new BLE advertising data is received, even if they don't have permission to send information to that device. This reveals the user's location, so we'll need a permission grant, and we may want to let users set bounds on the area or time in which the grant applies. We need to figure out if a site with full geolocation access should automatically get access to scan for beacons.

We also need to figure out whether it's useful to constrain scanning to tabs, or whether we need to figure out ServiceWorker access from the start. From the use case of walking around a museum listening to the sound corresponding to the closest exhibit, I'm inclined to think we have to figure out full background access from the start.

pzboyz, 2015-12-14:
Curious to know if you want to allow the website to register for any BLE advertising data or only 
for Eddystone formatted packets or for a specific Service UUID or ...?

How has this problem been solved for Wifi? Are websites able to list SSID and MAC addresses of nearby access points or has this been prohibited because of the fact this to some extents reveals their location?

jyasskin, 2015-12-14:
Any BLE advertising data, I think. The problem hasn't been solved for WiFi yet.

samgiles, 2016-02-01:
Mozilla have the [WiFi Manager API](https://developer.mozilla.org/en-US/docs/Web/API/WifiManager) although, signed package apps only for security/privacy reasons.  So not really web.

I can't articulate a sec issue with Eddystone beacons right now - but I believe some exist.

JosefJezek, 2016-02-04:
Please Scan API asap... ;-)
I can not read ID numbers of beacons from web app on background.

rodmaz, 2016-03-06:
Limiting the scan for a specific period or geographic area would be a very limiting specification IMHO. Why not giving control to the user which websites have requested permission to scan BLE advertisement and allow revoking that permission as they wish. Eddystone/iBeacon standards all allow an app to "subscribe" to a class of advertisements, so should it work with websites. 
An example: user visits a travel website and reviews about the city of Paris. He/she could subscribe to BLE advertisements and provided the city of Paris and has beacons through the city, users could subscribe to it.

VinceGa, 2016-03-23:
A serviceworker responding to a registered adv. event is a fantastic idea!

nemik, 2016-05-13:
I would just also like to chime in here that I think scanning for any advertisements would be best. For my use-case, I don't really need active scanning, only passive, but a complete implementation would ideally support active as well.

My particular use-case is I'm developing some very low-power battery-operated sensors that are non-connectable, and only transmit their status every so often using BLE advertisements. The small packet size of an advertisement means I can barely pack in my data in there so needing to specify additional things like a name or a service would just waste space for me. 
So being able to receive the raw advertisement data and parse it myself would be ideal.

jyasskin, 2016-05-13:
For advertisements, https://rawgit.com/jyasskin/web-bluetooth-1/scanning/scanning.html#bluetoothlescanfilter proposes to let you filter by manufacturer ID or the UUID of a Service Data field. What other way are you hoping to identify your device's packets? We probably can't give raw access to the bytes of the advertisement because of security worries, but most of the fields  should be available.

nemik, 2016-05-13:
Thanks! Being able to filter on manufacturer ID would be nice, I use that already. I was more concerned about only being able to filter on service UUID's or names which would mean I'd have to use more bytes in the advertisement for just that filtering. I currently don't use those to maximise the size of manufacturer-specific-data part of the packet which is where I shove in all my sensor info. Advertising packets are quite small and I'd like to use as much of them as possible for my data.

I am curious about what the security worries would be when reading the other bytes though? I'd really like to see the entire advertising packet and parse its contents how I'd like.

PR #239 (2016-08-02): Specify a Bluetooth Scanning API.
This doesn't cover scanning from a service worker yet.

Fixes #191.

Previews at https://rawgit.com/jyasskin/web-bluetooth-1/scanning/scanning.html and https://rawgit.com/jyasskin/web-bluetooth-1/scanning/index.html#bluetoothdevice
commit 9c56bc43 (2016-08-02): Specify a Bluetooth Scanning API. (#239)

* Specify a Bluetooth Scanning API.

The API can filter by name, namePrefix, serviceUUID, the ID in a manufacturer data entry, or the UUID in a service data entry, or can explicitly opt into receiving all advertisements.

This doesn't cover scanning from a service worker yet.

Fixes #191.
closed 2016-08-02 by jyasskin