whatwg/bluetooth #194: Add connect and disconnect gatt server Bluetooth events.

beaufortfrancois, 2016-01-06:
The same way, WebUSB added `connect` and `disconnect` events (https://github.com/reillyeon/webusb/commit/5806adfc523ff5146bc9fa883367b024d1d62699), it would be nice to have them in WebBluetooth so that Web Apps can detect connected/disconnected devices.

Just a proposal there...
```js
navigator.bluetooth.addEventListener('connectGatt', handleGattServer);
navigator.bluetooth.addEventListener('disconnectGatt', handleGattServer);

function handleGattServer(event) {
  var gattServer = target.value;
  console.log('Gatt Server connected: ', gattServer.connected);
});

beaufortfrancois, 2016-01-06:
Same as #159 
Sorry for duplicating issues....

jyasskin, 2016-01-27:
I'm initially inclined not to fire a 'connected' event when `gattServer.connect()` returns. Is there a reason we need it? (Obviously we need 'disconnected' since the page has no other way to discover that happened.)

beaufortfrancois, 2016-01-27:
I was thinking we could use for a "reconnect" scenario after a disconnect happened but... I'm not sure either we need it ;)

jyasskin, 2016-01-27:
Oh, right, I forgot auto-connection. Let's handle that by having `connect()` hang until a connection is made, with `disconnect()` being the way to cancel it. So, further discussion on #152. And #159 or #114 will handle the `disconnected` event.

closed 2016-01-27 by jyasskin