whatwg/bluetooth #199: Consider simplifying the 3-way mapping for allowed devices.

g-ortuno, 2016-01-15:
@noncombatant mention that the 3-way mapping is very complex and encourage us to consider a simpler approach:
This 3-way mapping system seems complex. I wonder if it could be simplified, such as by combining  the origin and the UUID?

    device-id := origin + "," + uuid

jyasskin, 2016-01-15:
Is this really a spec issue?

g-ortuno, 2016-01-15:
The spec very explicitly defines how the map should be structured:

`For each origin, the UA MUST maintain an allowed devices map, whose keys are the Bluetooth devices the origin is allowed to access, and whose values are pairs of a DOMString device id and an allowed services list consisting of UUIDs for GATT Primary Services the origin is allowed to access on the device.`

Having the implementation match the spec as closely as possible makes it easier to make changes in the future and helps keep track of where each feature is implemented.

jyasskin, 2016-01-25:
Specs are full of descriptions that don't match the implementations exactly. https://tc39.github.io/ecma262/#sec-map-objects has a good example, where it's specified in terms of a linear search to describe the external behavior, but also requires sublinear access. We shouldn't diverge where it doesn't make the specification simpler, but in this case, I think organizing the information a UA has to remember about a device into {whole-UA, per-origin, per-window} does make it easier to understand. I don't think it diverges too much from the implementation, either: even if you map from internal-id to web-visible-id, and then from that id to the origin, the allowed servers, and the internal-id, it's recognizable as the same information as the spec wants.

So I'm not inclined to change this in the spec unless there's a concrete proposal that makes it easier to understand.

closed 2016-01-25 by jyasskin