whatwg/bluetooth #204: Is it possible to get raw bytes of the BLE packet?

keremgocen, 2016-01-27:
Not sure if this is implemented yet. I was looking for something like the Android [scanRecord](http://developer.android.com/reference/android/bluetooth/le/ScanRecord.html) method to parse a custom ble format. This should speed up the adoption of custom formats such as Eddystone, is this something i can do at the moment?

jyasskin, 2016-01-27:
We're not planning to expose the raw bytes, but custom formats, including Eddystone and iBeacon, use the standard [manufacturer](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothadvertisingdata-manufacturerdata) or [service data](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothadvertisingdata-servicedata) fields, which we do expose. There's even an [example of parsing Apple's format](https://webbluetoothcg.github.io/web-bluetooth/#example-145d3a9c).

These aren't implemented anywhere yet, and they're not going to be as useful as they should be until we add a scanning API (to find all nearby beacons), which isn't coming until the next version of the API (after GATT communication ships).

closed 2016-01-27 by jyasskin

ritesh-dalal, 2016-05-24:
is device.watchadvertisements() function api available in the chrome for android or in chromium , i am using chromium version 53.0.22748.0 for android and getting 'device.watchAdvertisements is not a function' , i want to read the service data of the device founded in order to show eddystone beacons. Is there any way to scan for eddystone format beacons ? Please assist.

beaufortfrancois, 2016-05-24:
`watchadvertisements` is not implemented yet in Chrome.
Please check out https://github.com/WebBluetoothCG/web-bluetooth/pull/239 to learn more about how BLE Scanning will work in the future.

ritesh-dalal, 2016-05-24:
i am using chromium version 53.0.22748.0 and it seems it is not implemented in that , then is there any way to read the service data from the advertisements which is already implemented in the chromium ?

jyasskin, 2016-05-24:
No, it's not implemented yet.