whatwg/bluetooth #211: Discovery method

JosefJezek, 2016-02-02:
Could you add Discovery method? Like this startDiscovery method from Chrome...

https://developer.chrome.com/apps/bluetooth#method-startDiscovery

I would like to create Eddystone beacons scanner like this Android app...
https://play.google.com/store/apps/details?id=physical_web.org.physicalweb

beaufortfrancois, 2016-02-02:
It is already there. Learn more about this at https://developers.google.com/web/updates/2015/07/interact-with-ble-devices-on-the-web and check out samples at https://googlechrome.github.io/samples/web-bluetooth/device-info.html and more

g-ortuno, 2016-02-02:
I think he's referring to the scanning API #191

beaufortfrancois, 2016-02-02:
You're right.

JosefJezek, 2016-02-02:
@g-ortuno Yes, thank you :+1: 

We have also lib for node.js, check out this link
https://github.com/sandeepmistry/node-eddystone-beacon-scanner

jyasskin, 2016-02-02:
Yeah, it's still a duplicate of #191.

closed 2016-02-02 by jyasskin