whatwg/bluetooth #215: Consider an event when a device is physically disconnected

jyasskin, 2016-02-24:
Calling [`BluetoothRemoteGATTServer.disconnect()`](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothremotegattserver-disconnect) disconnects the current Realm and allows the UA to physically disconnect if no other Realm is connected. [`gattserverdisconnected`](https://webbluetoothcg.github.io/web-bluetooth/#eventdef-bluetoothdevice-gattserverdisconnected) in turn reports if the current Realm was disconnected without asking to be, but this doesn't say whether the device was [revoked](https://github.com/WebBluetoothCG/web-bluetooth/pull/210) or the physical connection was lost (and I think we want to preserve that ambiguity).

@beaufortfrancois reports that some devices treat physical disconnection as a semantic actions, and change their state in response, and web pages want to convey that to the user. Is that beyond the BT4.0 behavior that peripherals can have only one physical connection?

We could add a `.physicallyConnected` attribute to the `gattserverdisconnected` event, and use that to distinguish Realm vs UA disconnections (treating the UA as including its OS/platform here). I don't think we want to let a Realm force the UA to disconnect, since that can interfere with other Realms, but the information leak from telling a Realm whether other Realms are connected seems small enough to accept.

Thoughts?

scheib, 2016-02-24:
The use case I heard, a site wanting to know when a peripheral disconnects,
seems to be satisfied with a gattserverdisconnected message without more
information.

If a disconnection happens at any general time, the site can explain the
connection was lost, and prompt to reconnect. However, when the site
performs an operation that it expects will result in a disconnection upon
completion, it can wait for gattserverdisconnected and then show a
confirmation.

The chance that a false disconnection and confirmation will occur due to
other reasons seems low.

jyasskin, 2016-02-24:
@beaufortfrancois Does the GATT operation cause the remote device to disconnect, or does the remote device wait until the local device physically disconnects before applying the queued GATT operations?

beaufortfrancois, 2016-02-25:
For DFU case, the GATT operation (characteristic write) causes the remote device to disconnect as you can read at http://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk52.v0.9.2%2Fbledfu_appswitching.html&cp=4_0_2_4_2_2_2,

> First, you must write the value 0100 to the CCCD for the DFU Control Point characteristic to enable DFU notifications. Next, you must write the value 01-XX (where XX is the code for the type of image, see DFU Control Point) to the DFU Control Point to start a DFU procedure. **The application will then disconnect and reset into bootloader/DFU mode.** Note that no acknowledgement will be sent when writing to the Control Point, but the application sends a GAP disconnect on the link to tell the peer that the link is closed. You can then transfer the firmware image.

[image]

beaufortfrancois, 2016-02-25:
Adding @thegecko who might be interested.

beaufortfrancois, 2016-02-25:
And @mjdietzx as well

thegecko, 2016-02-25:
I agree with @scheib, assuming the ```gattserverdisconnected``` event is fired in all cases, I can't think of a use case when the user/developer needs to know why the disconnection happened.

For the DFU case, the disconnect is part of the process so won't come as an unexpected event, being considered in the flow of the code.

Is there a possibility here that multiple ```gattserverdisconnected``` events are raised due to different reasons?

jyasskin, 2016-02-25:
@thegecko Right now, the spec only says to fire one, when the current Realm is disconnected "surprisingly", which happens either when the user revokes permission or when the device physically disconnects, but not when the Realm calls `.disconnect()`. (I could make `.disconnect()` also fire the event without changing anything significant.)

This issue suggests to fire a second kind of `gattserverdisconnected`, when the device physically disconnects. We'd only actually see that if the user calls `.disconnect()`, since revocation would block it. But it sounds like we don't actually need this, since the user calling `.disconnect()` isn't a part of the protocol.

thegecko, 2016-02-25:
IMO this event _should_ fire when the Realm calls ```.disconnect()```. It offers a single place to do any clean up / reconnection no matter the cause of the disconnect.

kenchris, 2016-03-09:
I agree with @thegecko - this is what I was expecting to happen when I was writing a demo app, so I guess that I might not be the only one

beaufortfrancois, 2016-03-09:
@kenchris It has been added: https://github.com/WebBluetoothCG/web-bluetooth/pull/219

jyasskin, 2016-03-09:
Yep. And ... it sounds like we don't actually need a physical-disconnection event that would fire some time after the page calls `.disconnect()` when the device is actually disconnected from the UA. We just need to implement the already-specified `gattserverdisconnected` event for physical-disconnection-while-the-page-is-logically-connected.

I'll close this issue, but please re-open it if you come up with a use case.

closed 2016-03-09 by jyasskin