whatwg/bluetooth #220: startNotifications should always write descriptor if device is not bonded

beaufortfrancois, 2016-03-01:
If devices are not bonded, notifications won't be remembered after a disconnection.

Therefore, instead of giving the responsibility the check whether devices are bonded or not to the UA, I suggest `startNotifications()` should always write descriptor no matter what.

Same applies to `stopNotifications()` of course.

WDYT?

Background: https://bugs.chromium.org/p/chromium/issues/detail?id=589796#c9

mjdietzx, 2016-03-01:
I think this is necessary as well. I have a use case for this where we are doing a firmware transfer over the air, the link is lost for some reason, and the first thing we need to do is re enable notifications on a characteristic before starting the transfer where we left off.

Whether we always write the descriptor no matter what, or we actually read it and only write it if it is not enabled is up to you (i can't think of any problems with either way but the second seems a bit cleaner).

g-ortuno, 2016-03-10:
@jyasskin: I looked at this a bit closer. We should be clearing the [`active notification context set`](https://webbluetoothcg.github.io/web-bluetooth/#active-notification-context-set) upon disconnection.

jyasskin, 2016-03-10:
There are some characteristics (Service Changed comes to mind, although it's not user-visible) that queue up information to send to bonded devices, and then dump it out on re-connection. If we make sites re-register for notifications/indications on re-connection, they'll generally miss that set of notifications. @Vudentz also makes that point at the bottom of http://permalink.gmane.org/gmane.linux.bluez.kernel/66966. That said, it's going to be a fair amount of bookkeeping in the UA to keep the queue of notifications that a site hasn't seen. Maybe we should make them re-retrieve the data.

The spec definitely should say that non-bonded devices clear their notification set on disconnection.

beaufortfrancois, 2016-03-14:
What should we do then?

g-ortuno, 2016-03-14:
For the non bonded case we just have to wait for @jyasskin to fix the spec.

PR #279 (2016-09-06): Clear the active notification context set on disconnection.
Fixes #220.

Preview at https://api.csswg.org/bikeshed/?url=https://raw.githubusercontent.com/jyasskin/web-bluetooth-1/clear-notifications-on-disconnection/index.bs#clean-up-the-disconnected-device.
commit d8538378 (2016-09-06): Clear the active notification context set on disconnection. (#279)

Fixes #220.
closed 2016-09-06 by jyasskin