whatwg/bluetooth #221: Read/Write attribute by handle

VinceGa, 2016-03-23:
GATT is the abstract layer on top of ATT. A service may have multiple instances of the same characteristics. I suggest we change the operation on characteristics to the attribute level(get/set by handle) instead of by service/UUID.

jyasskin, 2016-03-23:
We do support having multiple instances of the same characteristic through methods like [`getCharacteristics(uuid)`](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothremotegattservice-getcharacteristics), but because no OS exposes operations on the attribute level, we can't insist that browsers do so. It wouldn't be implementable. We don't provide IDs for the multiple instances because Apple's CoreBluetooth doesn't provide them (e.g. [`CBCharacteristic`](https://developer.apple.com/library/ios/documentation/CoreBluetooth/Reference/CBCharacteristic_Class/index.html)), but the browser should make the objects distinguishable with `===`.

closed 2016-03-23 by jyasskin

VinceGa, 2016-03-23:
On Android, can we use BluetoothGattCharacteristic.getInstanceId()? Not sure if the handle is exposed in CoreBluetooth though. IMHO all GATT operations should be mapped to ATT level. More importantly, we need to separate GAP from GATT.(GATT.connect does not make sense from the core spec's perspective :-)). I understand Android/BlueDroid stack is implemented this way. But they are changing/refactoring...

jyasskin, 2016-03-23:
Android, BlueZ, and I think Windows all have persistent instance IDs, but since Mac/iOS doesn't, it's hard to expose it in a spec that needs to be implemented across all of them. These instance IDs also don't expose the actual 16-bit attribute handle in general, and can't be used to look up a characteristic by its handle.

If the core Bluetooth spec intends all of its users to call ATT operations directly, why would it provide the GATT procedures in 3.G.4? It seems like they're there specifically so we don't need to map all GATT operations to the ATT level.

[`device.gatt.connect()`](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothremotegattserver-connect) is defined in terms of the GAP Connection Establishment procedures. It's spelled "gatt.connect()" because it connects to a GATT Server, not some other type of server or client that might also be accessible via GAP. (This is hard to follow in the core spec because 3.C.9.3.8.2 just says "The Host shall initiate a connection." without linking to any definition of that in terms of lower level features.)

Overall, it looks like the core spec has one or two unnecessary abstraction layers (ATT and maybe GAP), and Web Bluetooth follows Android, iOS, BlueZ, and Windows in omitting them from the user-facing interface. Even if the layers are actually necessary for users, we'd need to get the platforms fixed before Web Bluetooth could depend on them, since browsers are implemented on top of the platforms.

VinceGa, 2016-03-23:
My understanding of GAP is that it separates the link management from the data operations. One example is the introduction of central and peripheral roles. After US called device.gatt.connect(), it can also be a GATT server who's providing data. This can enable a lot of use cases. Just my $0.02.