whatwg/bluetooth #222: Support for custom profiles.

VinceGa, 2016-03-23:
We need to add support on custom profiles/services/characteristics. More and more applications are using the custom profiles instead of the adopted ones. So all the references to the services/characteristics should be by UUID instead of by names.
Thanks.

jyasskin, 2016-03-23:
The names are just shortcuts to the UUIDs through the algorithms in https://webbluetoothcg.github.io/web-bluetooth/#standardized-uuids. It's totally possible to pass a UUID anywhere a name is accepted. If there's a piece of code you can't write but want to, could you re-open this with that example?

closed 2016-03-23 by jyasskin

VinceGa, 2016-03-23:
There are 2 types of UUIDs: 16-bit that adopted by SIG (standardized)and 128-bit that everyone else can use for his/her own services/characteristics. I could not find the API to query by my own 128-bit UUID. Any hints?

jyasskin, 2016-03-23:
For example, https://github.com/WebBluetoothCG/demos/blob/gh-pages/bluetooth-toy-bb8/index.html#L281 calls `gattServer.getPrimaryService("22bb746f-2bb0-7554-2d6f-726568705327")` with a 128-bit UUID.

VinceGa, 2016-03-23:
Yep, that's exactly what I needed. Thanks a lot!