whatwg/bluetooth #223: Chrome on Windows 10

VinceGa, 2016-03-24:
We may want to leverage the newly added Bluetooth APIs in Windows 10 to enhance the pairing support in Chrome on Windows.
Examples are here:
https://github.com/Microsoft/Windows-universal-samples/tree/master/Samples/DeviceEnumerationAndPairing

Let me know if you need help.

jyasskin, 2016-03-24:
Thanks for the link, and we do intend to take advantage of the latest APIs on each platform, but this repository is about the specification, not the Chrome implementation. Discussion of the Chrome implementation belongs in https://groups.google.com/a/chromium.org/forum/#!forum/web-bluetooth, not here. Chrome bugs can go in https://crbug.com/new following the instructions at https://www.chromium.org/developers/how-tos/file-web-bluetooth-bugs.

closed 2016-03-24 by jyasskin

beaufortfrancois, 2016-03-29:
Thank you @VinceGa!
This is very valuable. I'll make sure this repository is shared with the people working on the implementation of Bluetooth on Windows.