whatwg/bluetooth #224: Not working on Xiaomi Mi 3 with Android 4.4 (KitKat) version

akki10, 2016-03-25:
I have tried the following sample on my Mi3:

https://googlechrome.github.io/samples/web-bluetooth/device-info.html

Its showing the following error, even when my bluetooth is turned on:
Requesting Bluetooth Device...
Argh! NotFoundError: Bluetooth adapter not available.

I have also enabled the web-bluetooth flag in the chrome browser but the result is still the same.
With same steps its working on Nexus 5 with Android Marshmallow.

Is it the Android version problem or something else?

thegecko, 2016-03-25:
Check your chrome version is the latest dev (marshmallow) or a side-loaded nightly build (lollipop) and the ble experimental flag is switched on.

akki10, 2016-03-25:
Thanks for the fast reply thegecko. Yes, my chrome app is updated and flag is also enabled. Is it the problem with Android version? Because it KitKat.

g-ortuno, 2016-03-25:
Yes. KitKat is not supported. The BLE APIs in KitKat are different than the APIs in L and M, which are the APIs Web Bluetooth uses.

scheib, 2016-03-25:
... and the reason Chrome does not attempt to be backwards compatible and
use multiple APIs is because M is the first version that allows runtime
permission requests. Older versions of android would block users from
updating Chrome without accepting a new up-front Bluetooth permission --
and any upgrade friction to Chrome is a security risk for users.

jyasskin, 2016-03-25:
Also, this isn't the forum to report bugs with Chrome's implementation. Use https://crbug.com/new for that, or ask questions at https://groups.google.com/a/google.com/forum/#!forum/web-bluetooth.

closed 2016-03-25 by jyasskin

beaufortfrancois, 2016-03-30:
@akki10 If you can upgrade your phone to Android Lollipop, you may want to have a look at http://stackoverflow.com/questions/34810194/can-i-try-web-bluetooth-on-chrome-for-android-lollipop