whatwg/bluetooth #225: Device Discovery: suggest to add "Allow duplicate" in the BluetoothScanFilter

VinceGa, 2016-03-30:
I suggest we add a filter option for discovery event to filter out duplicate advertising packets. We should enable applications to respond to either the first received adv packet or continuous packets from the same device. In some use cases, the application is only interested in the first received adv packet from the device. In some other cases, it wants to get all the adv packets with maybe changing RX values.

Those filters are exposed from both core bluetooth and BluetoothLeScanner.
It's highly recommended that we have granular filter options to 1)improve performance/response time. 2)be more energy efficient.

Thanks.

Reference:
http://developer.android.com/reference/android/bluetooth/le/ScanSettings.html
https://developer.apple.com/library/ios/documentation/CoreBluetooth/Reference/CBCentralManager_Class/index.html#//apple_ref/doc/constant_group/Peripheral_Scanning_Options

jyasskin, 2016-03-30:
Note that [discovery](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetooth-requestdevice) doesn't produce "events"; it fills a dialog from which the user can pick devices. The browser has to pick whether to receive the first or all advertising packets from each device, and we're finding [use cases](https://crbug.com/599056) that require us to aggregate information from multiple packets in order to decide whether to include a device in the dialog.

We are planning a Scanning API, which will need this sort of control over which advertising packets are delivered.

jyasskin, 2016-08-29:
I believe [`BluetoothLEScanOptions.keepRepeatedDevices`](https://webbluetoothcg.github.io/web-bluetooth/scanning.html#dom-bluetoothlescanoptions-keeprepeateddevices) accomplishes this.

closed 2016-08-29 by jyasskin