whatwg/bluetooth #231: Beacon Broadcast Support

kahjav, 2016-04-26:
It would be fantastic for web-bluetooth to allow a web page to broadcast a bluetooth beacon, specifically a physical web beacon (Eddystone URL). This would work well with #191 - allowing web pages to talk to each other locally through browsers and URL's 

This could be achieved through filling an advertisement, or specifically broadcasting an Eddystone URL packet.

apptum, 2016-07-03:
Being able to configure a device to advertise in peripheral mode via a web page would open up a whole world of opportunities ( @kevinahuber: that's what you describe, isn't it?):
Possible use case:
- People in a train station looking for a ticket could get informed that there are others with the same destination nearby and be suggested to get a group ticket.
- While reading an online article in a cafe the user could get asked if she wants to broadcast what she's reading to find similar minded folks.

Assumed there will be more and more devices be able to act in peripheral mode in the near future [http://stackoverflow.com/questions/26482611/chipsets-devices-supporting-android-5-ble-peripheral-mode](url) spam could get a real issue though.

But the sound of webbluetooth + instant apps + the physical web is quite promising :)

related topic: [https://github.com/google/physical-web/issues/662](url)

@jyasskin: Do you guys may already have a roadmap for implementation?

beaufortfrancois, 2016-07-18:
As mentioned in https://webbluetoothcg.github.io/web-bluetooth/#introduction, 
> 
The first version of this specification allows web pages, running on a UA in the Central role, to connect to GATT Servers over either a BR/EDR or LE connection.
I'm quite positive we may want to expose a Web Bluetooth API to support broadcasting mode in the future. In the mean time, you should have a look at https://github.com/google/eddystone/tree/master/libraries/javascript/eddystone-advertising, an Eddystone Advertising Library that exposes simple functions that developers can use to advertise a Valid Eddystone packet from their Chrome OS device.

jracle, 2016-07-25:
Should we also classify in [beacons](https://github.com/WebBluetoothCG/web-bluetooth/labels/beacons) label ?

jracle, 2016-07-26:
thanks @jyasskin!

geekuillaume, 2017-09-21:
Is there any news regarding this feature request? Or somewhere else to follow the discussion around it?

kevinahuber, 2017-09-26:
@geekuillaume I like your excitement! What are your thoughts about it?

geekuillaume, 2017-09-27:
I think it would be a nice addition to the Web Bluetooth API. There is already a [Nearby API](https://developers.google.com/nearby/messages/overview) included on smartphone since Android 2.3 and on iOS.

Basically it would mirror the Java/Swift API (publish, unpublish, subscribe, unsubscribe) to broadcast or scan for around beacons in the form of a JS Buffer.

This API would requires a special permission from the user when first used by the page.

kevinahuber, 2017-09-27:
@jyasskin @beaufortfrancois I noticed this has been added to the implementation status and there is a [spec](https://webbluetoothcg.github.io/web-bluetooth/scanning.html). Has any progress been made on implementation? What would be the best way to kick that off?

beaufortfrancois, 2017-09-28:
The team is currently focusing on Web platform tests and Windows support. There is no progress on this yet.

geekuillaume, 2017-09-29:
@beaufortfrancois Is there something I can do to help define and integrate this feature?

scheib, 2017-10-06:
@geekuillaume Though seemingly a small feature, this is a large task.  I like it, but it's not something we're planning to do for a long time.  To do this would require:
- Specifying the API and getting review.
- Building support from security & privacy teams.
- Implementing Android, ChromeOS, Windows, macOS support -- some of which will be difficult (e.g. ChromeOS has peripheral mode support intentionally limited due to single-mode only adapters out there).
- Tests
- Implementing from the platform abstraction up to the web bindings, cross process. 
- Web Platform Tests
- Shipping, with friction unless you've also gained support from other implementations.

TomasHubelbauer, 2020-01-07:
Is this the best issue to subscribe to for notifications about progress on this feature? I'd love to make use of this when and if it becomes a part of the Web Bluetooth feature set.

scheib, 2020-01-18:
This is one good place to monitor.  

We're a long way off still from this.  Scanning has some partial implementation in place, and until that is completed I don't think we'll see any movement here.

Chromium's https://www.chromium.org/teams/web-capabilities-fugu project is partner focused and is tracking requests and interest in the chromium issue tracker.  Beyond expressing support here, you can also do so on related issues there.  Asking for the minimum required, with a non-speculative use case, will help the most.

https://crbug.com/897312 - Scanning
https://crbug.com/1043409 - Advertising only
https://crbug.com/1018925 - Peripheral (Advertising, Services, Characteristics)

jimmywarting, 2021-02-08:
I would really like this, there is not anything close enought to https://developers.google.com/nearby that lets u hook up two devices with webrtc and start sharing files locally without the help of a server

samjgorman, 2022-01-28:
Just voicing my support for this feature as well. I'm quite interested in enabling peer discovery with web bluetooth and then allowing for a flexible choice of transport layers to exchange this info over a p2p connection...

getify, 2022-10-16:
I'm building several apps (installed PWAs) that run only on your local device, with no server.

These apps aspire let you share your data with the same app on another device, using a variety of peer-to-peer techniques, so that a server never has to be involved.

The current best zero-server sharing is via animated QR codes and a device camera. That's a bit awkward/annoying, but it works. Other options I'm exploring include webaudio+mic, [webnfc] (https://github.com/w3c/web-nfc/issues/563), etc. WebRTC DataChannel would also be nice (when internet is available), but only if signaling could be done peer-to-peer (using something like webbluetooth or webnfc).

For example, I have a clipboard/copy-paste app to copy text on one device, then have that text available to paste on another device. Right now, that app encrypts your data, saves it to a gist via your github acct, then lets the other device pull it. I'm adding the animated QR stuff right now.

But I would super love to have webbluetooth P2P to offer much easier/more-ergonomic data sync between devices (without any internet).