whatwg/bluetooth #232: After gatt.connect() server coming back undefined.

danhklein, 2016-04-26:
After .requestDevice, I'm unable to get server out of next promise. Would love advice about what is happening. Thanks in advance!

navigator.bluetooth.requestDevice({
              filters: [{
                namePrefix: ['BB']
              }]
            })
            .then(device => {
           //I am getting device object here
     
             return  device.gatt.connect();
        })
        .then(server => {
        //server coming back undefined.
           server.getPrimaryService("22bb746f-2bb0-7554-2d6f-726568705327");
        })

g-ortuno, 2016-04-26:
Thanks for reaching out! In the future please file any issues regarding the implementation in [crbug.com/](https://bugs.chromium.org/p/chromium/issues/entry). I've opened [crbug.com/606988](http://crbug.com/606988) for you and included a couple of questions there so we can figure out the problem.

closed 2016-04-27 by scheib

beaufortfrancois, 2016-04-27:
@danhklein Did you see a picker like one below?
[image]