whatwg/bluetooth #236: FR: Remove startNotifications/stopNotifications methods

beaufortfrancois, 2016-05-04:
I wonder if we could get rid of `startNotifications`/`stopNotifications` methods and only keep the add/remove EventListener on `characteristicvaluechanged`.

 `startNotifications`/`stopNotifications` methods are usually combined with the event listeners and I think it would make developers life easier to only have to take care of event listeners.

```js
navigator.bluetooth.requestDevice({filters: [{ services: ['heart_rate'] }]})
.then(device => device.gatt.connect())
.then(server => server.getPrimaryService('heart_rate'))
.then(service => service.getCharacteristic('heart_rate_measurement')
.then(characteristic => {
  characteristic.addEventListener('characteristicvaluechanged', onHeartRateChanged);
});

thegecko, 2016-05-04:
Would there ever be a situation where you want to pause (stop) notifications without having to remove all the event handlers?

jefBinomed, 2016-05-04:
I totaly agree, and you will use a 

```javascript
characteristic.removeEventListener('characteristicvaluechanged', onHeartRateChanged);
```

to be an equivalent of ```stopNotifications``` 

It could be clear and helfull for developpers 

so for me +1

beaufortfrancois, 2016-05-04:
@thegecko Pausing notifications would simply mean removing the `characteristicvaluechanged` event handler. It doesn't seem that bad as you can re-add it when you want to resume notifications.
Which situation are you thinking of?

thegecko, 2016-05-04:
I was playing devil's advocate if multiple handlers were added :)

beaufortfrancois, 2016-05-04:
😈

kenchris, 2016-05-04:
How does that fit with what Generic Sensor API is doing? @tobie @riju

tobie, 2016-05-04:
See @domenic's [comment](https://github.com/w3c/sensors/issues/88#issuecomment-193306303) on this topic and @annevk confirming adding listeners shouldn't have side effects in https://github.com/w3c/sensors/issues/88#issuecomment-194873093.

This has additional benefits for us that we can pass sensors around without paying the cost of starting them up.

g-ortuno, 2016-05-04:
Keep in mind that notifications are not the only thing that can trigger the `characteristicvaluechanged` event, reading a characteristic also triggers the event. It's possible for someone to want to perform an action based on this event but not want to subscribe to notifications.

beaufortfrancois, 2016-05-04:
Argh... you're right @g-ortuno. I forgot about that.
With that and the fact that adding listeners shouldn't have side effects, I believe we can close this "feature request".

closed 2016-05-04 by jyasskin