whatwg/bluetooth #237: requestDevice() not considering already connected devices

martijnthe, 2016-05-04:
A device could already be connected to the machine that the UA runs on, prior to the `requestDevice()` call. When a connection is established, devices often stop advertising, so won't get found by scanning and won't be "found" by `requestDevice()`.

A connection could have been established, prior to the `requestDevice()` call, for various reasons:
- if another site has already connected to it,
- the platform or another app running on it connected to it,
- the device connected to the machine that the UA runs on (in case the machine was acting as a peripheral at the same time).

Looking at the [steps that the UA has to take](https://webbluetoothcg.github.io/web-bluetooth/#ref-for-dom-bluetooth-requestdevice-3) for `requestDevice()`, it seems that already-connected, non-advertising devices are ignored and will not be shown in the list. I think the UA should also match already-connected devices, using the GATT cache and possibly also the last known advertisement data to match.

beaufortfrancois, 2016-05-09:
For info, Chrome current implementation uses GATT cache when using `requestDevice`.

jracle, 2016-07-25:
Hi @martijnthe, 

reading @beaufortfrancois's answer, and from my experience with actual implementation, `requestDevice()` returns a mix of connected and advertizing devices.

So shall we close this specific issue?

jyasskin, 2016-07-25:
It's an issue with the spec, not the implementation. I agree the spec should say to include connected-but-not-advertising devices in the list to filter.

martijnthe, 2016-07-25:
+1 to what Jeffrey said.

jracle, 2016-07-26:
@jyasskin OK I see, cause I'm a bit confused with what `requestDevice` returns.

I'm used to using APIs which return only connected peripherals, and offering other methods / or filter for returning devices that are advertising.

I might add other issues / questions that are obvious for you sorry :). I'm re-discovering the spec..

PR #266 (2016-08-05): Include connected devices in scan results.
Fixes #237.

Preview at https://api.csswg.org/bikeshed/?url=https://github.com/jyasskin/web-bluetooth-1/raw/requestdevice-consider-connected/index.bs#scan-for-devices

@martijnthe, see if the wording looks right.
commit 8a8aedb3 (2016-08-05): Include connected devices in scan results. (#266)

Fixes #237.
closed 2016-08-05 by jyasskin