whatwg/bluetooth #238: Missing API for GATT Write w/o Response

martijnthe, 2016-05-04:
There does not seem to be a way to specify that the GATT Write w/o Response operation should be used when calling `writeValue()`.

I don't think this should be "abstracted away" / left up to the UA implementation because this is the only GATT operation by which GATT clients can have more than one GATT packets in flight towards the GATT server. The other write operations have to be ack'd before the next write can be sent out, making it very slow.

Most applications that I've seen that need to optimize for throughput (Pebble smartwatch, Parrot's drones, ...), usually use GATT Write w/o Response in combination with GATT Value Notifications for data in the other direction.

Also related: https://github.com/WebBluetoothCG/web-bluetooth/issues/26

jyasskin, 2016-05-04:
Do you have devices that allow both Write with Response and With w/o Response on the same characteristic?

What behavior do you want when the requested write type isn't among the characteristic's properties?

I suspect the API shape will be `writeValue(value, {type: 'without-response'})`

martijnthe, 2016-05-04:
> Do you have devices that allow both Write with Response and Write w/o Response on the same characteristic?

Is possible and allowed by the BT spec.
I'm sure someone will come up with a good use case for having both enabled for the same characteristic.

> What behavior do you want when the requested write type isn't among the characteristic's properties?
I'd say if `with-response` was explicitly specified, error out the way you would do normally when an operation is attempted that isn't allowed. If it's not explicitly specified, fallback to write with response?

> I suspect the API shape will be writeValue(value, {type: 'without-response'})
Sounds good to me!

360fun, 2016-08-03:
Hi, I think I may got stuck in this issue: basically I'm trying to cache the GATT server connection, but when I write some data it executes the command and it disconnects right after! So the second time that I try to write a command using the cached variable, I get the error "Uncaught (in promise) DOMException: GATT Service no longer exists.".
The trick seems to re-connect every time the GATT server, but this adds latency and this don't allow applications like driving drones, cars, robots...since they need a real time feedback! :(

I started to guess why this happens and then I read that the kind of command in this case, for my device, is "without response", so, guessing, when the API doesn't get anything after the promise, it disconnects the GATT server. If this is right...is there a workaround or we need to wait an update in the API? :\

g-ortuno, 2016-08-03:
It would be great if you could open a Chromium issue with your problem. Here are the instructions:

https://www.chromium.org/developers/how-tos/file-web-bluetooth-bugs

beaufortfrancois, 2016-08-03:
For info, @360fun also reported this issue at https://github.com/poshaughnessy/web-bluetooth-parrot-drone/issues/1#issuecomment-237092958

360fun, 2016-08-03:
Yes I did :) I'm kind of lost...but now I've a new way to investigate the issue, thanks! :D I never worked with Bluetooth before, so is good to learn! ;)

beaufortfrancois, 2016-08-04:
We'll continue conversation at https://github.com/poshaughnessy/web-bluetooth-parrot-drone/issues/1#issuecomment-237092958 to not pollute this thread and come back if it's a Web Bluetooth API issue.

Emill, 2016-09-16:
I'm also in favor for adding this option.

From the Bluetooth spec:
> This sub-procedure is used to write a Characteristic Value to a server when the client knows the Characteristic Value Handle and the client does not need an acknowledgement that the write was successfully performed.

It basically informs that Write Without Response should be chosen when the client does not need an acknowledgement. In this case I guess the "client" should refer to the application (and not the stack or browser) since the application is the only one who knows if an acknowledgement is really needed or not. The Bluetooth spec does _not_ state that Write Without Response can only be used if no other write sub-procedures are allowed according to the characteristic's permissions.

If the client wants to write many values and it ignores Write Responses, and the browser by itself chooses Write With Response, it results in wasteful significant delays.

360fun, 2016-09-16:
In the end I found that the problem was in my device: a watchdog is killing the connection after 500ms, so I just have to use a loop to keep alive the connection! :) But was nice to discover this thing anyway and know a bit more about Bluetooth. :D

Vudentz, 2016-09-19:
> If the client wants to write many values and it ignores Write Responses, and the browser by itself chooses Write With Response, it results in wasteful significant delays.

It goes both ways, if the remote device does not support this mode for some reason then the values would be ignored so at least it shouldn't be the default behavior.

beaufortfrancois, 2016-09-20:
Just to be clear, this option would only be useful if one single bluetooth characteristic supports `Write w Response` AND `Write w/o Response` at the same time. In this case, UA may choose the bad one by default I agree.
If not, UA will always choose the right option, the one that is specified by the writable bluetooth characteristic.

bluetooth-mdw, 2018-01-31:
@beaufortfrancois I have a device with a characteristic which supports both write and write without response. The device is the BBC micro:bit. It has an event bus which is exposed over Bluetooth via input and output event characteristics. For the input event, you can use either write or write without response. The point is that events are a generalised mechanism for communicating with the micro:bit and the decision as to whether acknowledged or unacknowledged writes are required has to be taken on an application by application basis. We definitely need this.

timmyhadwen, 2018-05-24:
Hi All,

Also need this for Silicon Labs Bluetooth over the air updates which have both Write and Write without response. Any updates on whether this can be added in the near future?

Cheers
Tim

reillyeon, 2018-05-25:
@timmyhadwen, can you clarify whether you need support for both methods because of your application logic or because the UA is choosing the wrong one (perhaps the device advertises support for both and one is broken)?

poshaughnessy, 2018-06-28:
On this topic (I hope), please may I check... It looks from [the spec](https://webbluetoothcg.github.io/web-bluetooth/#dom-bluetoothremotegattdescriptor-writevalue) that `writeValue` does not return a response; it says that in the successful case, the Promise will be resolved with `undefined`? 

(We're developing a Bluetooth peripheral and it should be returning a success response from the 'write' command, but I'm not receiving it back...)

I wondered if this is by design and if so, how this would fit in with this split between write _with_ response, and write _without_ response. Should write _with_ response return a value from the Promise?

bluetooth-mdw, 2018-12-24:
I'm wondering if this is this ever going to be addressed? 

It's clearly a mistake in the Web Bluetooth specification. Characteristics may support both WRITE (requests) and WRITE WITHOUT RESPONSE and a browser cannot be expected to second guess which is the most appropriate to use for a given application. Only the application developer can decide whether reliability or responsiveness are their priorities. There need to be two distinct APIs available or a flag which allows developers to indicate which variant should be used. You can continue to default to the current auto (not very) magic approach for backward compatibility. 

There are plenty of examples of characteristics with both write operations supported, clearly documented at bluetooth.com.

beaufortfrancois, 2019-01-02:
@reillyeon @dougt Shall we add `WriteWithResponse` and `WriteWithoutResponse` methods to the Web Bluetooth Spec?

Emill, 2019-03-18:
BlueZ now finally supports to select which write type to perform (Write With Response or Write Without Response) since https://git.kernel.org/pub/scm/bluetooth/bluez.git/commit/?id=fa9473bcc48417d69cc9ef81d41a72b18e34a55a. This means it is now implementable on all platforms, so there shouldn't be any reason to hide the option for the user anymore.

As we have seen, there are multiple devices out in the market allowing both write types and it would be good to fully support them.

jyasskin, 2019-03-18:
I'm not implementing this and don't know when @reillyeon's team will have time, but I'd probably design the API as, `writeValue(data, {response: "with"|"without"|"guess"})`, although I haven't thought through the exact strings in that options struct.

Emill, 2019-03-18:
I think I would rather use `auto` instead of `guess`, since it doesn't guess anything. It picks whatever is possible according to the characteristic properties.

MikeTheDane, 2019-04-11:
I suggest the following:
writeValue(data) for backward compatibility. Does whatever currently is used.
writeValue(data, true) for GATT Write
writeValue(data, false) for GATT Write Without Response

bluetooth-mdw, 2019-04-11:
@MikeTheDane 's suggestion looks good to me.

beaufortfrancois, 2019-04-11:
@reillyeon Do you think you'd have time to implement this? If not, I'm happy to have a look and start prototyping something.

beaufortfrancois, 2019-04-12:
@MikeTheDane, @bluetooth-mdw, @Emill, @poshaughnessy, @timmyhadwen, @martijnthe What do you think of this proposal?
If that looks good to you, I'll send a PR to update the spec for @reillyeon, @jyasskin and @g-ortuno to have a look.

```js
// Write value to a BLE characteristic. Client trusts UA to know which type to send.
try {
  await myCharacteristic.writeValue(someData, { response: 'auto' });
  // Same as our current `myCharacteristic.writeValue(data);`
} catch(error) {
  // Error is thrown if BLE characteristic does NOT support any write operation.
}

// Write value with response to a BLE characteristic.
try {
  await myCharacteristic.writeValue(someData, { response: 'with' });
} catch(error) {
  // [NEW] Error is thrown if BLE characteristic does NOT support writeWithResponse.
}

// Write value without response to a BLE characteristic.
try {
  await myCharacteristic.writeValue(someData, { response: 'without' });
} catch(error) {
  // [NEW] Error is thrown if BLE characteristic does NOT support writeWithoutResponse.
}
```

Emill, 2019-04-12:
Yes that looks ok to me!

reillyeon, 2019-04-12:
I think the advantage of `"auto"`, `"with"` and `"without"` is that they somewhat map to the language in the spec but in isolation I feel like `writeValue(someData, { response: 'with' })` will be hard for a reader to interpret. Does it mean that the response is "with"? As an alternative can I suggest `"optional"` (the current behavior, `"required"` (write with response), and `"never"` (write without response)?

g-ortuno, 2019-04-12:
What about making a separate function for WriteWithoutResponse? It feels weird to return a promise for an operation without a response.

thegecko, 2019-04-12:
> It feels weird to return a promise for an operation without a response

This is quite common, the `write` functions for example.

Emill, 2019-04-12:
I think if a user would specify the write type rather than leave it out, he/she would have good knowledge of the Bluetooth spec and the different options already. I though have to admit that "response: with" sounds a bit weird so I am open to other naming.

About receiving a promise even for write without response, that might be good for flow controlling. For example, on Android you need to wait for the callback even if you use write without response because it will wait before delivering the callback in case the buffers are full.

MikeTheDane, 2019-04-12:
I like the terms suggested by @reillyeon. They seem pretty intuitive. But a separate function would be optimal (I agree with @Emill that a promise is definitely applicable to GATT Write w/o response)

beaufortfrancois, 2019-04-15:
We could also decide to add two new methods `writeValueWithResponse` and `writeValueWithoutResponse`.

```js
// Write value to a BLE characteristic. Client trusts UA to know which type to send.
try {
  await myCharacteristic.writeValue(someData);
} catch(error) {
  // Error is thrown if BLE characteristic does NOT support any write operation.
}

// [NEW] Write value with response to a BLE characteristic.
try {
  await myCharacteristic.writeValueWithResponse(someData);
} catch(error) {
  // Error is thrown if BLE characteristic does NOT support writeWithResponse.
}

// [NEW] Write value without response to a BLE characteristic.
try {
  await myCharacteristic.writeValueWithoutResponse(someData);
} catch(error) {
  // Error is thrown if BLE characteristic does NOT support writeWithoutResponse.
}
```

reillyeon, 2019-04-15:
New methods seem most readable to me.

MikeTheDane, 2019-04-16:
I absolutely agree that two new methods would be the ideal solution.

beaufortfrancois, 2019-04-17:
@reillyeon and @MikeTheDane seem to agree.
@thegecko,  @bluetooth-mdw, @Emill, @poshaughnessy, @timmyhadwen, @martijnthe What do you think of this proposal?

Emill, 2019-04-17:
Yes I think it's more clear now.

martijnthe, 2019-04-17:
+1 for new methods

bluetooth-mdw, 2019-04-17:
Looks good to me :-)

thegecko, 2019-04-17:
No concerns

poshaughnessy, 2019-04-17:
I like the separate methods too. Looks intuitive and clear, thanks @beaufortfrancois

beaufortfrancois, 2019-04-18:
FYI I've just sent a PR to the spec: https://github.com/WebBluetoothCG/web-bluetooth/pull/433

reillyeon, 2019-07-10:
#433 has been merged. Implementation is tracked in Chromium [issue 672648](https://bugs.chromium.org/p/chromium/issues/detail?id=672648)

g-ortuno, 2019-07-10:
Would be nice if the comments I left on #433 were addressed

scheib, 2020-04-29:
This issue / patch to date has two outstanding areas I think need to be addressed before closing:

a) We should address g-ortuno's [comment](https://github.com/WebBluetoothCG/web-bluetooth/pull/433/files#r277496051) which requests writeValue() be specified specifically e.g. "If the Write bit is set in characteristic’s properties, use the Write Characteristic Value procedure, otherwise use the Write without Response procedure."

b) Perhaps Web Bluetooth should not be owning enforcing the algorithm selection based on Characteristic Properties.  That is, if the application requests writeWithoutResponse we should attempt it by sending the request to the operating system without verifying Characteristic Properties.  This question was raised by @dlech after discussing with @Vudentz on bluez patch thread [src/gatt-client: always check properties in WriteValue](https://marc.info/?t=158799982800006&r=1&w=2).  Apparently some devices may operate with an algorithm even though they don't mark those property bits.

scheib, 2020-04-29:
My personal thoughts on:

a) writeValue should be specified to follow a specific algorithm.  I agree with @g-ortuno's preferring writeWithResponse.

b) I believe Web Bluetooth should not check characteristic properties.  Applications can access characteristic properties to check themselves if needed.  If an algorithm can work with a device advertising incorrect property bits I'd like the application to have a chance to attempt it.

If the algorithm won't work it will fail regardless.  The value of having Web Bluetooth check would be i) consistent error type and ii) cross platform consistency.  But these come at the expense of application functionality.

MikeTheDane, 2020-04-29:
I don't think we should abstract the Web Bluetooth developer too far from
the Bluetooth specification. When you write an app, you should know which
properties are supported for each individual GATT characteristic that you
are interacting with. It is very easy to figure this out in the chrome
browser if you do not have access to documentation (
chrome://bluetooth-internals/#devices  ).

There should be 2 write methods (write w/o response and write w/ response)
in addition to the existing "best effort write" which should remain for
backward compatibility.

My two cents...

bluetooth-mdw, 2020-04-29:
I agree with @MikeTheDane . The remote device will enforce its own rules anyway.

odejesush, 2020-04-29:
Please correct me if I'm wrong, but it sounds like there's still agreement to have two explicit methods for write with/without response while keeping the original writeValue. This change was already added to the spec with #433. It also looks like there's agreement that the specified write should be done regardless of the characteristic properties. I also agree with these ideas.

If that's the case, what's left is to agree on a default algorithm for writeValue(), which @scheib and @g-ortuno prefer writeWithResponse(). Would this algorithm also be the "best effort write" @MikeTheDane and @bluetooth-mdw ?

MikeTheDane, 2020-04-29:
I would leave the "best effort write" exactly as is - whatever that is. It
should only remain to ensure backward compatibility, and it should
potentially come with a note to not use it in new projects - perhaps even
be removed from the documentation entirely (it should have never existed in
the first place).Changing the method will break implementations. I should
note that in one of my implementations the GATT server device offered both
WriteNoResponse and WriteWithResponse for a characteristic used for
firmware upgrade over the air. Unfortunaltely writeValue() in my case
picked WriteWithResponse and the firmware upgrade took about twice as long.
Changing the GATT options to only offer WriteNoResponse fixed this issue,
but I could only do that because I am in control of the device firmware
(probably unlike most people).

g-ortuno, 2020-04-29:
The issue with best effort is that it's not specified, so what we do depends on the underlying platform. I think it would be good to specify what we mean by best effort.

MikeTheDane, 2020-04-30:
Giovanni, your short description "depends on the underlying platform" with
the added recommendation not to ever use it for that very reason should
suffice. Do you agree?

odejesush, 2020-04-30:
For not using `writeValue()` correct? It would still be good to explicitly state what `writeValue()` will do so that its behavior is consistent cross-platform, even if it is discouraged from being used.

dlech, 2020-04-30:
FWIW, the current implementation of `writeValue()` in chromium is already inconsistent across platforms, so "fixing" it could break existing users. For example, Android currently prefers write without response while other platforms prefer write with response if both corresponding write properties are set on a characteristic. Another example, BlueZ will give a NOT_SUPPORTED error when attempting to write if no write property flags are set while other platforms will raise a NOT_PERMITTED error.

odejesush, 2020-04-30:
That's one reason that the specification should declare what algorithm `writeValue()` should use. That would clear up any confusion on how this API is implemented for both users and implementors.

As for the error, the algorithm for write value currently specifies that in that scenario, `NotSupportedError` should be thrown. So it looks like that's a bug for Chromium's implementation using BlueZ.

MikeTheDane, 2020-05-01:
The Bluetooth specification specifies that the DEVICE should throw the
NotSupportedError error over the link when an unsupported GATT request is
received. It does not specify, nor should it specify, how that is
communicated to upper layers on the peer device or what action to take when
this error occurs.

I strongly believe that Web Bluetooth should specify writeValue() as an
obsolete method that should not be used in new implementations due to the
inconsistent behavior on various platforms. Changing the behavior of
writeValue() in ANY way would completely remove the need to keep it - which
is to ensure backward compatibility. I would even suggest that the method
is retired after a certain period of time (leave it after this time, but
remove it from the documentation).

odejesush, 2020-05-04:
Ah, I misunderstood where the error was coming from. From a Web Bluetooth point of view, I was suggesting that the error returned should be consistent if the platform does not support writing to a characteristic without the write property set. I do acknowledge that this might not be easy to do given that different errors are produced by different platforms.

I see what you mean now, and I agree now that `writeValue()` should be kept as is and eventually become deprecated in favor of the more specific write operations.

reillyeon, 2020-05-06:
I agree that `writeValue()` should be deprecated in favor of the new methods. I also agree with removing enforcement that the device supports the requested algorithm in order to provide broader compatibility. We could add a (one time) console warning when using an unsupported method but let the request proceed.

scheib, 2020-05-29:
Interested web-devs: 

@dlech 's chromium patch adding `writeValueWithoutResponse` etc landed via https://crrev.com/c/2159770.

Give it a try in Chrome Canary with chrome://flags/#enable-experimental-web-platform-features set.

reillyeon, 2020-10-07:
The new writeWithResponse() and writeWithoutResponse() methods are available in Chrome 85.

closed 2020-10-07 by reillyeon