whatwg/bluetooth #242: How to handle specific iOS Characteristic Properties

beaufortfrancois, 2016-06-06:
According to https://developer.apple.com/library/ios/documentation/CoreBluetooth/Reference/CBCharacteristic_Class/#//apple_ref/c/econst/CBCharacteristicPropertyNotifyEncryptionRequired, there are 2 Characteristic Properties available in iOS 6.0 (not MacOS) that are not specified in bluetooth. Can we describe in the spec what should happen? 

```
CBCharacteristicPropertyNotifyEncryptionRequired
Only trusted devices can enable notifications of the characteristic’s value.
Available in iOS 6.0 and later.

CBCharacteristicPropertyIndicateEncryptionRequired
Only trusted devices can enable indications of the characteristic’s value.
Available in iOS 6.0 and later.
```

I think `CBCharacteristicPropertyNotifyEncryptionRequired` should be treated as `CBCharacteristicPropertyNotify` and `CBCharacteristicPropertyIndicateEncryptionRequired` as `CBCharacteristicPropertyIndicate`. What do you think?

cc @jeromelebel

jyasskin, 2016-06-06:
The spec is based on the Bluetooth standard, rather than any particular platform interface, so I don't think it can say what should happen with non-standard property values.

Do you know what causes those to be set? Are they maybe just for building a peripheral, and not used on the central side? When they're set, are the CBCharacteristicProperty{Notify,Indicate} values also set?

jeromelebel, 2016-06-07:
According to the tests we did, the "EncryptionRequired" suffix gives the information that a pairing will be triggered if the characteristic is used.
So it can definitely be converted to a regular notify and indicate properties in Chromium.

jyasskin, 2016-06-07:
I'm really curious what BLE protocol messages MacOS uses to get that information before actually trying to write the descriptor.

jeromelebel, 2016-06-07:
In general, is there a way to know if using a characteristic will need a pairing or not?

jyasskin, 2016-06-07:
@jeromelebel BT4.2 3.F.3.2.5 says not. You just try to communicate and get "Insufficient Encryption" or "Insufficient Authentication". (3.C.10.3)

AndreyErmoshin, 2016-06-08:
>  so I don't think it can say what should happen with non-standard property values.
they are not platform specific properties , it is only names of properties from iOS/MacOS SDK. These properties used to identify characteristics which can be accessed only in encrypted way. So, for example if iOS application has central role, then after discovering characteristics from peripheral - iOS app. can determine if encryption is required for that characteristic or it can be read/written/subscribed without bluetooth protocol handshake needed for encryption.

jeromelebel, 2016-06-08:
So we did some test between iOS and OS X. A characteristic on the peripheral set as CBCharacteristicPropertyNotifyEncryptionRequired is received as CBCharacteristicPropertyNotify. So Apple is not able to send the suffix "EncryptionRequired" to the central. And it can only be used only on the peripheral side.

beaufortfrancois, 2016-06-08:
We can close this then.
Thanks everyone!

closed 2016-06-08 by beaufortfrancois