whatwg/bluetooth #243: Add a "connectable" filter to requestDevice

beaufortfrancois, 2016-06-14:
It would be nice to either add a "connectable" filter to requestDevice to only show Bluetooth Devices that are connectable

```js
let options = { filters: [{ services: [0x1234], connectable: true }],
                optionalServices: ['battery_service'] };
navigator.bluetooth.requestDevice(options)
.then(device => device.gatt.connect())
...
```

or make `requestDevice` calls only match Bluetooth Devices that are connectable as the intent behind `requestDevice` seems to connect to the GATT Server. The non-connectable bluetooth devices would be handled with `requestLEScan`.

jyasskin, 2016-07-27:
What should we do on Android, where we can't detect connectability? We could say that the UA MAY treat non-connectable devices as not matching, but we'll definitely have platform divergence and confused developers.

Do we have lots of services that both connectable and non-connectable devices advertise? Or is this just something we're running into because we're trying to write generic applications that target a more varying set of devices than most of our developers will target?

jracle, 2016-07-28:
I'd rather put the `boolean connectable` as a nullable property in BluetoothDevice (see #253).

mkv123, 2018-08-20:
FYI: Since API Level 26 (Android 8.0), Android does have the information available in its scan results.